Klipper3d/klipper #2169: Feature request: PROBE_ACCURACY for X and Y to test endstop accuracy

vladbabii, 2019-11-11:
It would be nice to be able to test accuracy of endstops. 

I did this so far by editing config files and replacing x/y with z, but it's a lot of time, very error-prone and getting quite complex when you have multiple z steppers.

Maybe name it ENDSTOP_ACCURACY ?

Thank you

Hywelmartin, 2019-11-12:
For greater endstop repeatability try this 
https://github.com/KevinOConnor/klipper/blob/064804b68867a83fd77f687b2f504069b74de771/config/example-extras.cfg#L354

KevinOConnor, 2019-11-12:
As indicated, one can use `ENDSTOP_PHASE_CALIBRATE` as a mechanism for determining the accuracy of the endstop (as well as a mechanism for tuning endstop phases) - https://www.klipper3d.org/Endstop_Phase.html

-Kevin

vladbabii, 2019-11-12:
```
It is not valid to use this feature on the XY endstops of corexy printers (as the XY position is not determined by a single stepper on corexy kinematics).
```

My printer is a corexy.

KevinOConnor, 2019-11-15:
Okay.  It's possible to get at the info via the GET_POSITION command.  Otherwise, a developer will need to implement and submit this.

-Kevin

KevinOConnor, 2019-12-23:
I'm going to mark this as inactive.

-Kevin

closed 2019-12-23 by KevinOConnor