Klipper3d/klipper #2171: One motor for all extruder

Akex2, 2019-11-12:
Hi
I have a issue with conf file but not just that : 
My printer has 4 extruder but just one motor for all
It's my old prototype for uderstand how it work : 
https://www.youtube.com/watch?v=AvobluGq-zE

https://www.youtube.com/watch?v=yvdoWnII3OA

But i have a issue in conf file because i "used multiple times" the same pin. eh yes ;) 
But it's ok for me !
I search this error and go out. ;)
I do that : 
[image]

Klipper boot it's good! yes... but the strange think is now : 
- when i use only one extruder in print is ok(1 or 2 or 3 ...).
- but when i use 2 or more:
    the first extruder work well
    but after the first change the extruder go in a wrong way ! 
    and after that the extruder keep this and the wrong way ( extrude= retract - retract= extrude)
    i tryed to retake the first extruder taked in the print and it turn also in a wrong way

After that i do that : 

[image]

But with this modification klipper take the conf of the last extruder in the config file, and never change it( and i need to change the step_distance, the drive gear didn't have the exact same diameter)

I tryed to find the good way in toolhead.py
[image]
If i good remember, comment 418 ligne and the wrong way go out, but is the same, the extruder never change and never change the setup for step distance.

Can you help me to fix that and maybe add a option for one motor used multiple times.
Thanks

[printer.cfg.LOG](https://github.com/KevinOConnor/klipper/files/3837591/printer.cfg.LOG)
[klippy.LOG](https://github.com/KevinOConnor/klipper/files/3837592/klippy.LOG)

KevinOConnor, 2019-11-13:
> My printer has 4 extruder but just one motor for all

Interesting.

> But i have a issue in conf file because i "used multiple times" the same pin. 

It's not valid to disable those pin checks.  If one creates multiple extruders in the micro-controller it will just lead to chaos.

It would be necessary for the host code to create just one stepper object, and then share it among multiple extruders.  (Similar to how the shared_heater code works.)

-Kevin

vladbabii, 2019-11-14:
@Akex2 your issue with reversed direction might be because you do a retract as the last move (asumption on my part). Can you do a small extude before changing extruders ?

Akex2, 2019-11-18:
Thx all for yours reply
Ok @KevinOConnor 
I try do some change but is not easy to understand how to do that properly
I need just shared the pin not other in stepper.
I do that, but i think is not enought,  help is really apreciated 
[image]
Thx

aschor, 2020-04-05:
Hi, is there a workaround for now ? I have a printer with hotswappable tool head, so I want to be able to use same pins for motor and sensors , but with different values for nozzle_size and pid values ... 

I can if I change the config manually each time I switch the toolhead but ...... that's not quite user friendly...

ld3300, 2020-04-13:
Okay, I think I have figured out roughly what needs to happen, but it will take me a while.  I'm assuming we want to try and also include the option to have the second instance reverse driven compared to the first, as in #2517.  I'm slow, so don't hold your breath.

EzDzO, 2020-11-11:
Hi, I have a different setup, but the same problem. 

I have a single extruder motor and a single heater but 2 different nozzle sizes that I can change using a stepper motor. 

Basically I need to create two extruder objects that have the same heater, the same sensor , the same pid values and the same stepper motor, but two different nozzle sizes and pressure advance settings.

Do you have any updates?

KevinOConnor, 2020-11-13:
> Basically I need to create two extruder objects that have the same heater, the same sensor , the same pid values and the same stepper motor, but two different nozzle sizes and pressure advance settings.

In this situation, just define one extruder in the config file and create `T0` and `T1` macros that alter the pressure_advance setting (via the `SET_PRESSURE_ADVANCE` command).  The nozzle_size in the config is just used for internal sanity checks - set the config to the largest possible nozzle size you will be using.

-Kevin

EzDzO, 2020-11-17:
> In this situation, just define one extruder in the config file and create `T0` and `T1` macros that alter the pressure_advance setting (via the `SET_PRESSURE_ADVANCE` command).  The nozzle_size in [...]

Ok, that makes sense.

Thank you

KevinOConnor, 2021-03-05:
I'm going to mark this as inactive.  It would be a nice feature to add, but it doesn't look like anyone is currently working on it.

-Kevin

closed 2021-03-05 by KevinOConnor