Klipper3d/klipper #2172: Permission denied etc/udev/rules

TheRunnerPi, 2019-11-13:
Hi :)

related to this ticket https://github.com/KevinOConnor/klipper/issues/835 i am trying to use this new rule, for an restart on connection:
```
## rule to restart klipper when the printer is connected via usb
SUBSYSTEM=="usb", ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="7523", ACTION=="add", RUN+="/b 
```
Making a new rule under /etc/udev/rules.d/ leeds to permission denied:

[image]

How can i get permission to that root?

Or how else can i copy the new rule to that root?

Regards,
Daniel

FHeilmann, 2019-11-13:
you're probably logged in as a non-root user, which means you don't have permissions to write to that directory. 

Not really a klipper problem, you need to log in with superuser credentials using your tool, or use SSH on your pi and escalate to root privileges using `sudo su`

TheRunnerPi, 2019-11-13:
Thank you @FHeilmann 

closed

closed 2019-11-13 by TheRunnerPi