Klipper3d/klipper #2175: PID Extrusion Scaling

ade, 2019-11-14:
Feature request / idea
Not sure if this is already a thing, had a look at src and it was not immediately obvious that it is.

Klipper should look at E movement and compensate the heating algorithm accordingly, in order for very fast extrusions to not cool the nozzle too much. This would mostly be noticeable with very large nozzles or high printing speeds.

In Marlin there is an experimental setting PID_EXTRUSION_SCALING for this.

KevinOConnor, 2019-11-15:
It's not implemented currently.  It's certainly possible to implement it if a developer is interested in working on it.

-Kevin

KevinOConnor, 2019-12-13:
It doesn't look like this is under active development, so I'm going to mark this ticket as inactive.

-Kevin

closed 2019-12-13 by KevinOConnor

KevinOConnor, 2019-12-30:
In retrospect, this looks similar to the existing #573. 

-Kevin