Klipper3d/klipper #2182: Yet another MMU support. some questions about gcode_macro

kakou-fr, 2019-11-16:
I have build a custom MMU2 witha skr-mini board. I currently use it with Marlin and a custom firmware.
I try to use it with klipper, and so i started to define a config file for it.
Actually I can home it, load/unload filament ... but in order to improve my config, I have some questions (I added the definition of my stepper in the post) : 

1) How could I get the status of an endstop from a gcode_macro.
For example, I need to unload the filament if the PINDA on the MMU is TRIGGERED, so I need to test the status from the endstop of [manual_stepper drive_stepper]
I have also a IR sensor on the extruder that detect that the filament is inserted, and need to get its status in order to pause the print if the filament is not correctly inserted in the extruder.

2) I need to define two endstop for the [manual_stepper drive_stepper], the PINDA that detect the insert of the filament, and an IR on the extruder that detect that the filament is correctly inserted.
but how to define two endstop (maybe min and max) in the config ?

3) the loading process need to move two stepper at the same time (the gear on the mmu and the extruder) in order to insert the filament (and stop when the IR is triggered). but how to move two stepper at the same time ?

4) in order to work with slic3r ... without modification, I need T0 T1 ... T4 gcode. It's possible to define a macro T that use the second character as an parameter ?

`
# based on https://github.com/mwr666/klipper/blob/sunbeam2.0c_multi_mcu_mmu2/config/printer-protodev-corexy-multi-mcu-mmu2.cfg

[manual_stepper drive_stepper]
step_pin: mmboard:PC5
dir_pin: mmboard:PB0
enable_pin: !mmboard:PC4
# 152
step_distance: .0065788
velocity: 20
accel: 10
endstop_pin: ^mmboard:PC2

[manual_stepper idler_stepper]
step_pin: mmboard:PB13
dir_pin: mmboard:PB14
enable_pin: !mmboard:PB12
# 25
step_distance: .040000
velocity: 100
accel: 80
#endstop_pin: ^mmboard:P1.27

[manual_stepper selector_stepper]
step_pin: mmboard:PC6
dir_pin: mmboard:PC7
enable_pin: !mmboard:PB15
# 80 step/mm
step_distance: .0025
velocity: 35
accel: 100
endstop_pin: !mmboard:PC0
`

[klippy.log](https://github.com/KevinOConnor/klipper/files/3854496/klippy.log)

kakou-fr, 2019-11-17:
I write a quick patch for manual_stepper.py for 1), but maybe not the better

```
    def get_status(self, eventtime):
         endstops = self.stepper.get_endstops()
         print_time = self.printer.lookup_object('toolhead').get_last_move_time()
         return {'endstop': ["open", "TRIGGERED"][not not endstops[0][0].query_endstop(print_time)]}
```

so i can test the status in macro

`    {% if printer["manual_stepper drive_stepper"].endstop == "TRIGGERED" %}`

kakou-fr, 2019-11-17:
If someone is interested, my current config (with a lot of macro en {% if ...%} is here https://github.com/kakou-fr/klipper/blob/master/config/printer-mmu-remix.cfg

vladbabii, 2019-11-17:
@kakou-fr 

You can query state of a gcode_button

```
[gcode_button danger_zone]
pin: helper1:ar22
press_gcode:
   M118 DangerZone off
release_gcode:
   M118 DangerZone ON
```

and in G-code macro
```
  {% if printer["output_pin dangerous"].value == 0 %}
  M118 Homing - skipping backing off due to not dangerous
  {% else %}
```

and use this to force  reading the value when printer starts up
```
[delayed_gcode welcome]
initial_duration: 1
gcode:
  M118 Hello and welcome!
 QUERY_BUTTON button=danger_zone
```

Hope this helps

vladbabii, 2019-11-17:
for (2) you can use a relay to connect to a single pin two endstops by wiring the endstop output through the relays - I recommand pair of solid state relays - https://images-na.ssl-images-amazon.com/images/I/613fIzTrA2L._SX425_.jpg

vladbabii, 2019-11-17:
for (3) i used a relay to connect step pin on step stick between 2 relays.

vladbabii, 2019-11-17:
(4) T0... Tx map to extruders. You can define extruder, extruder1, ... extruder5 and each T0-T5 will have activate_gcode and deactivate_gcode that can be run. You can fake multiple extruders by wiring the step/dir pins together on the step stick to the other step or you can remove check in klipper code that makes sure a pin is not reused.

kakou-fr, 2019-11-17:
Finally i put the IR pin to an stepper without endstop, so i can ask it state.

the last issues I have is :
- how to PAUSE a macro, because PAUSE pause octoprint, but the macro is interpreted to the end of the gcode_macro
- how to run two motor at the same time

thks for help

vladbabii, 2019-11-18:
Macros are interpreted at start of macro run. Pausing a macro will make no difference if you change stuff. Maybe run chained delayed_gcode at each step and check if a gcode variable has some special value and abort or just don't chain next delayed gcode?

kakou-fr, 2019-11-18:
something like PAUSE_NOW that PAUSE octoprint but also pause klipper (and so don't immediately  interpret the end of the MACRO)

KevinOConnor, 2019-11-18:
FWIW, this is something that I suspect adding a python module (eg, klippy/extras/mmu2.py) would be appropriate.  I suspect trying to implement the mmu semantics in macros will be difficult and fragile.

-Kevin

kakou-fr, 2019-11-18:
I already write the behavior as a set of MACRO. the only blocking point is the behavior of PAUSE that pause octoprint and not klipper, so it continue to execute some gcode after PAUSE.

>  I suspect trying to implement the mmu semantics in macros will be difficult and fragile.
Writing the macro was not so hard, the behaviour of the MMU is really simple. But why do you say that It will be fragile ?

kakou-fr, 2019-11-26:
I posted  a pull request with the two function i needed and an example of configuration for the mmu.

All is working without issue now

[image]
[image]

closed 2019-12-12 by kakou-fr