Klipper3d/klipper #2183: Problem with variable sending to start_print macro

macraj, 2019-11-16:
Hi, recently I tried to use start_print macro…
But not working correctly ( the variables which I pass to, seems not visible by macro )

Below start_print macro I’m using and gcode with variables:

[gcode_macro START_PRINT]
default_parameter_BED_TEMP: 59
default_parameter_EXTRUDER_TEMP: 210
gcode:
    # Start bed heating
    M140 S{BED_TEMP}
    M104 S{EXTRUDER_TEMP}
    # Use absolute coordinates
    G90
    # Reset the G-Code Z offset (adjust Z offset if needed)
    # SET_GCODE_OFFSET Z=0.0
    # Home the printer
    G28
    M109 S{EXTRUDER_TEMP}
    #G29 
    # Move the nozzle near the bed
    G92 E0
    G1 X0 Y0 F2000
    # Move the nozzle very close to the bed
    G1 Z0.15 F300
    # Wait for bed to reach temperature
    # M190 S{BED_TEMP}
    # Set and wait for nozzle to reach temperature
    
    G1 X60.0 E10 F600
    G1 X100 Z0.1 E12.5 F1000
    G92 E0
    G1 X150 F5000 

### gcode : ####
…
G90
M82
M106 S0
START_PRINT  T_BED={65} T_EXTRUDER={215}
; process PETG Optimal
; layer 1, Z = 0.188
…

And the printer is always start
 with default parameters ( 59 and 215 ).

[klippy.log](https://github.com/KevinOConnor/klipper/files/3854761/klippy.log)

Arksine, 2019-11-17:
The parameters in your gcode do not match the parameters defined in the macro (your gcode is using T_BED and T_EXTRUDER, the macro is using BED_TEMP and EXTRUDER_TEMP).

Also it appears that the slicer isn't generating the gcode correctly, as the values should not include braces.  It should look like:

```
START_PRINT BED_TEMP=65 EXTRUDER_TEMP=215
```

macraj, 2019-11-17:
Thx a lot ! 
Well it's so obvious when You look at someone solution. Shame on me for a stupid mistakes.
Again thx a lot.

closed 2019-11-17 by macraj