Klipper3d/klipper #2185: Saving Z offset from display

TwoToneEddy, 2019-11-17:
Hi,
Is it possible to save a Z offset from the display on the printer or is this impossible due to the SAVE_CONFIG command restarting the service? Just so i'm clear, should I find the best value for the Z offset then edit the printer.cfg file manually?

KevinOConnor, 2019-11-18:
I'm not sure what you are referring to with "z offset".  If you mean the probe z offset then I recommend running the PROBE_CALIBRATE command as described at https://www.klipper3d.org/Probe_Calibrate.html .  If you are referring to the z offset in SET_GCODE_OFFSET , then the recommended method is to define a START_PRINT macro and set the offset there - as described at: https://www.klipper3d.org/Bed_Level.html#determining-thermal-expansion

-Kevin

TwoToneEddy, 2019-11-18:
Sorry for the ambiguity, I'm used to marlin and my method of setting the distance between the nozzle and the bed is adjusting the bltouch z offset during a test print until I get the right amount of squish. In order to do it this way I thought there would be a way to tweak the offset from the LCD display (which I believe there is) and then save this to the config file some how. Is this not possible?

JohnEdwa, 2019-11-21:
This is asking basically the same as my issue #1170 was.

The proper procedure is still to figure it out using the Z-offset on the LCD Tune menu and then adding it to the config file as the proper BLtouch z-offset. It'll take a few tries to get it just right, and you can't change or save it on the printer itself.

The workaround is adding it as `SET_GCODE_OFFSET Z=x.xx` to your start G-code.   
I recommend adding it to Octoprints "Before print job starts" GCODE Script, as that is the easiest place to change it if need be as it doesn't require restarting Klipper or reslicing the file.

manu7irl, 2019-11-21:
Yes it is Possible,
Here a example of fragmented config files.
From Fl0r1s
As you can see in this menu.cfg file
There is a configuration to play with z-offset on the fly, this is like
babysteps in the M. Word firmware.
https://github.com/fl0r1s/cr10_klipper/blob/master/menu.cfg

KevinOConnor, 2019-12-13:
I'm going to close this ticket as it looks like the conversation has concluded.

-Kevin

closed 2019-12-13 by KevinOConnor