Klipper3d/klipper #2186: I don't understand why you don't calibrate delta correctly (size problem)

Nitrosito, 2019-11-17:
I know this is for code problems, but I do not understand why it does not calibrate well with the automatic methods ... I have introduced the measurement of the rods to the mm, I have done the DELTA_CALIBRATE and the extended method, but the 20x20 cubes measure 20.20 x 20.20 ... even if you repeat the extended method, keep measuring that ... I don't know what it can be (calibrated flow)

Any ideas? I repeat 3 or 4 times the calibration but same result :( allways 20.20 x 20.20

My config:
```

[stepper_a]
step_pin: P2.0
dir_pin: P0.5
enable_pin: !P0.4
step_distance: .00625
endstop_pin: !P1.25
homing_speed: 50
#angle = 211.042251
#arm_length = 339
#position_endstop = 393.149365

[stepper_b]
step_pin: P2.1
dir_pin: P0.11
enable_pin: !P0.10
step_distance: .00625
endstop_pin:  ^!P1.27
#angle = 329.315614
#arm_length = 339
#position_endstop = 393.985863

[stepper_c]
step_pin: P2.2
dir_pin: !P0.20
enable_pin: !P0.19
step_distance: .00625
endstop_pin: ^!P1.29
#position_endstop = 391.758346
#arm_length = 339
#angle:90

[extruder]
step_pin: P2.3
dir_pin: !P0.22
enable_pin: !P0.21
step_distance: .001204819
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: ATC Semitec 104GT-2
sensor_pin: P0.24
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: P2.5
sensor_type: Honeywell 100K 135-104LAG-J01
sensor_pin: P0.25
#control: watermark
min_temp: 0
max_temp: 110

[fan]
pin: P2.4

[mcu]
serial: /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00

[printer]
kinematics: delta
#   This option must be "delta" for linear delta printers.
max_velocity: 120
max_accel: 2000
max_z_velocity: 120
#   max_velocity for max_z_velocity.
minimum_z_position: -5
#delta_radius: 165.58115
#    delta_radius = smooth_rod_offset - effector_offset - carriage_offset

[delta_calibrate]
radius: 120
speed: 50
#horizontal_move_z: 5

[probe]
pin: !P1.28
z_offset: 0.1

samples: 3
[static_digital_output leds]
pins: P1.18, P1.19, P1.20, P1.21, P4.28

[mcp4451 stepper_digipot1]
i2c_address: 44
# Scale the config so that wiper values can be specified in amps.
scale: 2.25
# wiper 0 is X (aka alpha), 1 is Y, 2 is Z, 3 is E0
wiper_0: 1.0
wiper_1: 1.0
wiper_2: 1.0
wiper_3: 1.0

[mcp4451 stepper_digipot2]
i2c_address: 45
scale: 2.25
# wiper 0 is E1
wiper_0: 1.0

[bed_mesh]
bed_radius:110

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 53.229
#*# pid_ki = 0.648
#*# pid_kd = 1093.863
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 22.168
#*# pid_ki = 0.859
#*# pid_kd = 142.984
#*#
#*# [printer]
#*# delta_radius = 165.539905
#*#
#*# [stepper_a]
#*# angle = 210.150993
#*# arm_length = 338.115065
#*# position_endstop = 393.724659
#*#
#*# [stepper_b]
#*# angle = 329.304794
#*# arm_length = 341.706994
#*# position_endstop = 392.600184
#*#
#*# [stepper_c]
#*# angle = 90.000000
#*# arm_length = 340.943426
#*# position_endstop = 392.938009
#*#
#*# [bed_mesh default]
#*# points =
#*# 	-0.072822, -0.072822, -0.072822, -0.072822, -0.072822
#*# 	-0.018169, -0.018169, -0.010263, -0.035603, -0.035603
#*# 	0.040828, 0.010539, -0.006906, -0.037743, 0.149736
#*# 	-0.051773, -0.051773, -0.067410, -0.121982, -0.121982
#*# 	-0.058728, -0.058728, -0.058728, -0.058728, -0.058728
#*# x_count = 5
#*# y_count = 5
#*# min_x = -110.0
#*# max_x = 110.0
#*# min_y = -110.0
#*# max_y = 110.0
#*# x_offset = 0.0
#*# y_offset = 0.0
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*#
#*# [delta_calibrate]
#*# height0 = 0.1
#*# height0_pos = 62905.333,63039.333,62683.333
#*# height1 = 0.1
#*# height1_pos = 72659.333,72792.333,56566.333
#*# height2 = 0.1
#*# height2_pos = 61448.667,78535.667,61226.667
#*# height3 = 0.1
#*# height3_pos = 56969.667,71120.667,70764.667
#*# height4 = 0.1
#*# height4_pos = 60956.333,61090.333,73175.333
#*# height5 = 0.1
#*# height5_pos = 69494.000,57389.000,69272.000
#*# height6 = 0.1
#*# height6_pos = 75764.000,61296.000,60939.000
#*# distance0 = 64.96
#*# distance0_pos1 = 62599.617,63021.206,63073.186
#*# distance0_pos2 = 58296.482,67345.925,67469.465
#*# distance1 = 64.96
#*# distance1_pos1 = 62799.534,62622.299,63274.237
#*# distance1_pos2 = 61203.460,61005.492,71030.736
#*# distance2 = 64.9
#*# distance2_pos1 = 63203.935,62428.031,63073.186
#*# distance2_pos2 = 67643.303,58177.552,67469.465
#*# distance3 = 65.08
#*# distance3_pos1 = 63408.459,62630.227,62673.614
#*# distance3_pos2 = 71279.668,61124.070,61109.637
#*# distance4 = 65.06
#*# distance4_pos1 = 63205.950,63029.200,62475.071
#*# distance4_pos2 = 67677.694,67481.921,58218.418
#*# distance5 = 64.93
#*# distance5_pos1 = 62801.532,63225.932,62673.614
#*# distance5_pos2 = 61233.325,70967.680,61109.637
#*# distance6 = 65.04
#*# distance6_pos1 = 58631.711,66439.472,67209.906
#*# distance6_pos2 = 61517.559,60742.434,70696.600
#*# distance7 = 64.95
#*# distance7_pos1 = 61560.679,60787.337,70036.582
#*# distance7_pos2 = 67780.871,58291.732,66943.749
#*# distance8 = 64.95
#*# distance8_pos1 = 67382.222,58513.541,66555.597
#*# distance8_pos2 = 70940.855,61434.659,60842.061
#*# distance9 = 64.89
#*# distance9_pos1 = 70269.486,61472.451,60883.261
#*# distance9_pos2 = 67144.036,67611.944,58329.114
#*# distance10 = 65.03
#*# distance10_pos1 = 66749.053,67213.671,58551.211
#*# distance10_pos2 = 60961.712,70629.536,61420.317
#*# distance11 = 65.15
#*# distance11_pos1 = 61002.420,69971.039,61462.002
#*# distance11_pos2 = 58407.374,66822.019,67603.964
```
Thanks and regards

KevinOConnor, 2019-11-18:
> I have done the DELTA_CALIBRATE and the extended method, but the 20x20 cubes measure 20.20 x 20.20

An error of .20 is commonly the result of "filament swell" - that is expansion of the filament after it leaves the nozzle and/or thermal expansion of the plastic.  It is not indicative of a kinematics error, so you don't need to continue to perform delta calibration.

To correct this remaining error, you should double check your "e steps" - for example see: http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide#E_steps

If the e-steps are correct and you still have an error, then many slicer's will have an option to account for this type of error.  And/or it may be necessary to alter the printed model itself to account for the error.

-Kevin

Nitrosito, 2019-11-18:
Thanks for answering even if it's not a code problem. The truth is that I knew that function, but I have never liked to use it because I think that if everything mechanical is correct, I should drive the measures. If I don't get the perfect measurements, I will use that. Do you know someone who has needed to use it?

cheers and thank you again

cmock, 2019-11-21:
I regularily use the "horizontal expansion" setting in Cura when I need to print precise parts. I printed a test object with a series of holes in 0.1 mm increments and checked which one exactely fitted a rod of known size (a.k.a drill bit), and then set it accordingly, and now my parts fit together perfectly when designed with 0 clearance.

KevinOConnor, 2019-11-21:
Right.  Just to clarify, if one gets 20.200 for a 20mm cube and gets 202.000 for a 200mm cube then it is a kinematics error and would be fixed by extended delta calibration.  If one gets 20.200 for a 20mm cube and gets 200.200 for a 200mm cube then it is a "filament swell" issue and it can only be fixed at the slicer (or at object creation).

-Kevin

Nitrosito, 2019-11-21:
> Right. Just to clarify, if one gets 20.200 for a 20mm cube and gets 202.000 for a 200mm cube then it is a kinematics error and would be fixed by extended delta calibration. If one gets 20.200 for a [...]

Thanks kevin. I try this when i get a litle time. 
A lot of thanks!

closed 2019-11-24 by klipper-gitissuebot