Klipper3d/klipper #2187: Digital scales feedback connected via serial

spaxxomatic, 2019-11-17:
Hi Kevin 
I've branched out klipper a while ago [here ](https://github.com/spaxxomatic/klipper/tree/spi_mcu_support) and, for the purpose of using it for controlling a lathe, I've done the folllowing modifications:

1. MCU  is connected via SPI (was a hell of a work due to the master-slave nature of SPI)
2. Position feedback:  linear scales connected to Raspi via serial /dev/ttyAMA0

I was about to implement the positon feedback as an extension module, but I've hit some race conditons:  I tried to start the encoder polling via the serial device in a new thread. But, when the background thread started in serialhdi.py and my thread are both transferring data, I'm getting from time to time an EINTR signal. 
I'm not familiar with this signal, but if I understood right, the kernel sends it when a non-blocking write is active and another signal interrupts it. (https://unix.stackexchange.com/questions/253349/eintr-is-there-a-rationale-behind-it)
I assume that this happens when the SPI data transfer is active in the serialqueue thread, while the other thread is notified that serial data is avail and tries read it.  

This does not occur when I syncronize kernel calls by reading the position from the serial device in the same thread using the pollreactor in serialqueue.c. (see https://raw.githubusercontent.com/spaxxomatic/klipper/spi_mcu_support/klippy/chelper/serialqueue.c ) 

I'd like to keep things clean and separate the position reading from the MCU comm - so here are my questions: 
Do you ever had multiple poll reactors in different threads, polling multiple file handles? 
Do you need extra thread syncs when runnung multiple reactors? (mutex, etc)?
Did you encountered any issues with EINTR in this case? 
Or do you recommend to use only one (the main poll reactor from serialqueue) to watch all fd's  (via the pollreactor_add_fd) ?

KevinOConnor, 2019-11-18:
> Do you ever had multiple poll reactors in different threads, polling multiple file handles?

It's common to have multiple threads each using their own "struct pollreactor".  A single "struct pollreactor" is not designed to be used simultaneously from multiple threads.

> Do you need extra thread syncs when runnung multiple reactors? (mutex, etc)?

I'm not sure what is being asked here.  A single "struct pollreactor" is never used from multiple threads, so no synchronization is currently needed.

> Did you encountered any issues with EINTR in this case?

I haven't seen an issue with that in Klipper.  An EINTR in my experience results from using signal() and/or some very old profiling tools (which use signal() ).

> Or do you recommend to use only one (the main poll reactor from serialqueue) to watch all fd's (via the pollreactor_add_fd) ?

I don't understand the question.  There isn't a "main poll reactor" - every mcu gets its own "struct pollreactor" which is then only accessed from that mcu's serialqueue thread.

I'm guessing you've made a lot of changes to serialqueue.c for SPI and something has gone awry with that.

-Kevin

spaxxomatic, 2019-11-19:
> Do you need extra thread syncs when runnung multiple reactors? (mutex, etc)?

> I'm not sure what is being asked here. A single "struct pollreactor" is never used from multiple threads, so no synchronization is currently needed.

I understand that. My question was if the read/write ops of different threads need to be sync-ed. 
The poll reactors in different threads does not need sync, but they are listening on different file handles. In the picture below, the serialqueue has a nonblocking handle on the SPI device, and the thread sends data. The EINTR occurs when I'm writing data to SPI, and while writing, the other thread has data to receive and runs into the read routine. When he starts reading, the writing thread gets an EINTR error. 
But you might be right, I have to check, the problem could be caused by the fact I'm using a wrapper lib that registers a signal handler. 

[image]

spaxxomatic, 2019-11-19:
Yeah, the doc explains it http://man7.org/linux/man-pages/man2/epoll_wait.2.html
 

>  epoll_pwait()
>        The relationship between epoll_wait() and epoll_pwait() is analogous
>        to the relationship between select(2) and pselect(2): like
>        pselect(2), epoll_pwait() allows an application to safely wait until
>        either a file descriptor becomes ready **or until a signal is caught**.

So the writing thread should mask the signal by using the pthread_sigmask(SIG_SETMASK, &sigmask, &origmask) if he's not interested in the signal. 

BTW - I see there is a python implementation of the pollreactor (reactor.py), but the serialqueue implements its own reactor in C. Is this done for performance reasons?

spaxxomatic, 2019-11-19:
Just saw in the code that serialqueue.c is using poll, not epoll, vs the python implementation has three variants SelectReactor, PollReactor and EPollReactor, but the EPollReactor is not used.

cmock, 2019-11-21:
Just a wild idea: are you acquiring the serial device as your controlling terminal when opening it, and then maybe you get interrupts via a line discipline on it, like when a Ctrl-C is received?

KevinOConnor, 2019-11-22:
FYI, if there are questions on the current Klipper code, I can try to answer them.  Alas, I don't have any meaningful insight to code not currently in the main Klipper repo.

If you are looking for some high-level advice, I suggest getting everything working with the current uart/usb code first.  The sensor feedback loop is a lot of work, and doing that work with a new comm layer is, I suspect, going to add notable complexity.  Also, FWIW, I feel UART is both a simpler and superior protocol to SPI for the Klipper application.

-Kevin

spaxxomatic, 2019-11-23:
Yes, indeed the serial is a better choice for the way Klipper communicates, because it allows the MCU to send whenever it has time, while as SPI slave it must transmit on the speed and tempo given my the master. 
Nevertheless, it works fine now. The problem was that the scales read thread was using signals and not poll, while the MCU thread uses poll . Using both is not advisable (as Kevin hinted), because of some quirks in the way the Linux kernel is implemented. Some low-level reading here:
https://idea.popcount.org/2017-02-20-epoll-is-fundamentally-broken-12/
So thanks to all, from my perspective this issue is closed.

closed 2019-12-01 by KevinOConnor