Klipper3d/klipper #2189: Klipper vitual sd problem after printing

CaribDuck, 2019-11-18:
There is a problem after printing.

Octoprint can't use virtual sd normally.

The system keeps sending the error message "sd write not supported"

And I can't print using the Octoprint's UI.

Of course, I can print using the "m20, m23, m24" command on the terminal.

Before printing, the full name of the file is displayed in status.

But, after printing, only part of the file name displayed in status.

Did this affect the problem?

How can I fix it?

[image]   [image]

KevinOConnor, 2019-11-22:
It's not necessary to upload a print from OctoPrint to Klipper - once the virtual_sdcard config option has been configured properly, just uploading the file to OctoPrint is sufficient to print from Klipper.

When viewing the file listing in OctoPrint, it can help to click the "wrench icon" above the file listing, and select "only show files stored on SD".  After uploading a file to OctoPrint, it may also be necessary to manually refresh the file listing (by clicking the "refresh SD file" option and/or the "initialize SD" option in the "SD operations" menu - which is to the left of the "wrench icon").

Otherwise, this does not appear to be a Klipper issue, but an OctoPrint usability issue.
-Kevin

CaribDuck, 2019-12-01:
> It's not necessary to upload a print from OctoPrint to Klipper - once the virtual_sdcard config option has been configured properly, just uploading the file to OctoPrint is sufficient to print from [...]

I understand what you say.
I thought I had to use "upload to sd" to use virtual sd.
Thank you.

closed 2019-12-01 by CaribDuck