Klipper3d/klipper #2191: Pressure advance causing issues with narrow skin layers

oullah, 2019-11-19:
I am having an issue where if I lower my pressure advance, I get blobbing on the corners. But if I raise my pressure advance, I get bad underextrusion on skin layers which are narrow and the print head moves back and forth quickly. Is there any way to resolve this?

oullah, 2019-11-19:
Looks like I need to attach a log file. Is there somewhere else I can ask a general question like this about using Klipper?

JohnEdwa, 2019-11-21:
>Is there somewhere else I can ask a general question like this about using Klipper?

@KevinOConnor would prefer you to use [the Klipper mailing list or IRC channel](https://www.klipper3d.org/Contact.html), both cutting edge communication forms straight from the nineties.

KevinOConnor, 2019-11-21:
With the merge today of PR #1997 the pressure advance implementation has changed.  You'll want to retest with the latest version of the code.

I recommend tuning pressure advance using the documentation at https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md as it is difficult to tune pressure advance correctly from an arbitrary print.  If you continue to have issues with a tuned pressure advance, then you could try attaching photos of the prints that are showing the issue.

-Kevin

closed 2019-11-26 by klipper-gitissuebot