Klipper3d/klipper #2192: Jerk settings?

oullah, 2019-11-19:
I am seeing some waviness in the walls after corners that slowly dissipate. I think it's because I have a glass bed and it's causing vibrations to go through the belt. What settings can I use to reduce this issue? Is there anything I can do besides reducing acceleration? Seems like it's only an issue in the Y axis since my glass bed is heaving and puts a lot of strain on the belt.

jakep82, 2019-11-19:
Square corner velocity is the Klipper equivalent of jerk, although it's not exactly the same.

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L319

oullah, 2019-11-19:
Cool, but it's not intuitive whether I should raise it or lower it. If I lower it that means less shaking for the corners, but it also means more deceleration / acceleration as well.

jakep82, 2019-11-19:
Jerk does the exact same thing, although I believe the math is a little different.  Both settings control instantaneous velocity changes.

acid-maker, 2019-11-22:
What about delta kinematics?

I don't see similar setting for delta. There is only acceleration.

Hywelmartin, 2019-11-22:
@acid-maker 
just ad "square_corner_velocity: 5.0" in the [printer] section

acid-maker, 2019-11-22:
@Hywelmartin 

Great. Thanks - Maybe it would be good to add such section (practically copy from `example.cfg`) to `example_delta.cfg` file.

KevinOConnor, 2019-12-13:
I'm going to close this ticket as it looks like the conversation has concluded.

-Kevin

closed 2019-12-13 by KevinOConnor