Klipper3d/klipper #2193: abl not working

macraj, 2019-11-19:
Hi, my abl is not working correctly. I can see this when I am printing on  the entire bed... The left side is too high, middle is correct and the right side is too low.
I did screws tilt calibration, then calibrate Z-axis and then mesh bed levelling. All of that few times but it did not help... 
Tried to send G29 and nothing helps.
Is this my printers fault ?
I changed stock capacitive probe for inductive one, changed Z-axis nuts and what more can I do?
The Creality 'Melzi' board with TMS2208 doesn’t work in UART mode, but after 3-days fight (setting proper current in standalone mode) seems to be working now.
So how to check if abl works correctly ? Or how to configure abl to get it to work ?

[image]
[image]
[image]

BlackStump, 2019-11-19:
a suggestion change your G29 macro to be something like this
>[gcode_macro G29]
>gcode:
>    G28
>    BED_MESH_CLEAR
>    BED_MESH_CALIBRATE
>    BED_MESH_PROFILE SAVE=ENDER3 
 >   SAVE_CONFIG

change your  START_PRINT macro to something like this adding the bed mesh profile load
>[gcode_macro START_PRINT]
>default_parameter_bed_temp = 60
>default_parameter_extruder_temp = 180
>gcode = 
>	M104 S{180}
>	M190 S{60}
>	M140 S{BED_TEMP}
>	M104 S{EXTRUDER_TEMP}
>	G90
>	G28
>     BED_MESH_PROFILE LOAD=ENDER3
>	M109 S{EXTRUDER_TEMP}
>	G92 E0
>	G1 X0 Y0 F2000
>	G1 Z0.15 F300
>	G1 X60.0 E10 F600
>	G1 X100 Z0.1 E12.5 F1000
>	G92 E0
>	G1 X150 F5000

the ENDER3 you can change to what ever takes your fancy
on a side note you might want to read this
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print

macraj, 2019-11-20:
> a suggestion change your G29 macro to be something like this
>
>
> change your START_PRINT macro to something like this adding the bed mesh profile load
>
>
> the ENDER3 you can change to what ever takes [...]

Thx for the answer... I will change my G29 macro and start_print. 
I read about the clipper and tmc2208, but I can't find a pdn-uart to atmega connection. 
Probably someone made a mistake when designing the pcb and just connected the pull through the resistor. It remains only to replace the pcb or solder new paths.
For today, only a short print (less than 1 hour) is successful with pressure_advance enabled.
Although there is no problem with abl only with Creality or rather Chinese quality ...

closed 2019-11-20 by macraj