Klipper3d/klipper #2194: SKR Mini E3 V1.2 wrong temperature measurement

Magoo0876, 2019-11-21:
Hi,

I've played a bit with the new SKR mini E3 V1.2 board and liked it so far, but something was strange. After some time I've measured the head and bed temperatures, and they were too low. 
My sensors are 104GT-2 on the bed and B3950 on the hotted .

bed set to 60 measured 50
hottend set to 200 measured 160 

I've played with the pull-up resistor setting and found that 6k almost compensates the error. Also different sensor type settings did not do the expected. 

So I think the internal pulldown resistors of the STM32F102 are activated and this would cause a higher temperature reading for the software. I also tried to activate the pull-up resistors for the thermistor input pin, but this is not a possible setting in the config.

Thank's for your help.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3876934/klippy.log)

Magoo0876, 2019-11-23:
Room temperature during the test 23.8

Recv: ok FIRMWARE_VERSION:v0.8.0-132-gc188aa98 FIRMWARE_NAME:Klipper
Send: M105
Recv: ok B:26.2 /0.0 T0:26.3 /0.0

# measured thermistor 3950 102k2 measured pullup 4k69
Send: QUERY_ADC name=extruder PULLUP=4700
Recv: // ADC object "extruder" has value 0.952808 (timestamp 28.990)
Recv: // resistance 94893.790 (with 4700 pullup)
Recv: ok B:26.2 /0.0 T0:26.3 /0.0

# measured thermistor 104GT-2 102k6 measured pullup 4k69 
Send: QUERY_ADC name=heater_bed PULLUP=4700
Recv: // ADC object "heater_bed" has value 0.952686 (timestamp 53.790)
Recv: // resistance 94636.774 (with 4700 pullup)

I also tried the pullup setting in the config and ended up at 5250 to match at room temperature. @200set I got 165 at the hotend measured with meter and flir camera. As you see above, there is a needle in the haystack, the values differ a lot.

Thank's Kevin

Rubber1Duck, 2019-11-25:
Hi,
i have a problem similar to this toppic on an SKR Mini E3 DIP, but in the other direction!
Roomtemperatur between 21,8 °C.
Klipper shows up 16 °C both, Extuder an Bed!
[image]
For Testing i installed Marlin 2.0 and then with same Hardware ( board und termistors) 21 °C both extruder and Bed is showing up!
[image]
In Marlin i configured Sensor number 1 with ist the EPCOS 100K  termistor
in Klipper i configured EPCOS 100K B57560G104F (i tried all the other ones, but no one has the target of roomtemperatur)

Send: query_adc name=heater_bed pullup=4700
Recv: // ADC object "heater_bed" has value 0.969231 (timestamp 924.890)
Recv: // resistance 148050.000 (with 4700 pullup)
Recv: ok
[...]
Send: query_adc name=extruder pullup=4700
Recv: // ADC object "extruder" has value 0.969109 (timestamp 935.800)
Recv: // resistance 147446.245 (with 4700 pullup)
Recv: ok

Messurment termistor extruder 112k5, Bed 116k1 at the same temperature

whats going wrong there????

Magoo0876, 2019-11-25:
Hi, I'm happy that I'm not allone with this problem.
In the meantime I did the calibration for two sensor types connected on the SKR mini E3 1.2, as a workaroud and got the following valuse. 

[thermistor my_3950] 
temperature1: 23.6
resistance1: 95000
temperature2: 108
resistance2: 3750
temperature3: 198
resistance3: 390

[thermistor my_104GT2] 
temperature1: 23.6
resistance1: 95000
temperature2: 100
resistance2: 3750
temperature3: 200
resistance3: 300 

Further strange is the shape of the thermistor temperature curve during cooldown. There is a noticeable signal noise under 80°C.

[image]

KevinOConnor, 2019-11-26:
Odd.  If you change the `uint32_t aticks = 3;` on line 64 of src/stm32/adc.c to `uint32_t aticks = 4;` and then recompile and flash the micro-controller, does the behaviour change?

-Kevin

Rubber1Duck, 2019-11-26:
Hi Kevin,
i will test that right now.
Best Regards 
Peter

Rubber1Duck, 2019-11-26:
Hi Kevin,
i made this change and now the temperatures are to high
[image]

the resulting resistanc is to high

Recv: ok FIRMWARE_VERSION:v0.8.0-150-g43dfb3c9-dirty FIRMWARE_NAME:Klipper
Recv: ok
[...]
Send: query_adc name=extruder pullup=4700
Recv: // ADC object "extruder" has value 0.955311 (timestamp 333.600)
Recv: // resistance 100472.131 (with 4700 pullup)
Recv: ok
[...]
Send: query_adc name=heater_bed pullup=4700
Recv: // ADC object "heater_bed" has value 0.955311 (timestamp 378.190)
Recv: // resistance 100472.131 (with 4700 pullup)
Recv: ok
[...]
Send: query_adc name=heater_bed pullup=4700
Recv: // ADC object "heater_bed" has value 0.955311 (timestamp 389.590)
Recv: // resistance 100472.131 (with 4700 pullup)
Recv: ok
Send: query_adc name=heater_bed pullup=4700
Recv: // ADC object "heater_bed" has value 0.955311 (timestamp 392.290)
Recv: // resistance 100472.131 (with 4700 pullup)
Recv: ok
[image]

but the resistance (and adc value) is now stable ! (as you can see above), bevor it is going up and down more than 1000 ohm .......
This is the result for a SKR E3 DIP  !!!!!

Rubber1Duck, 2019-11-26:
The Steps at heating und cooling down are still there!
[image]
And again .... with Marlin Firmware the temeratures are right 
Best Reagards

KevinOConnor, 2019-11-27:
Very odd.  I don't see any issues with my "skr mini" board (though it's not in a printer).

Could you try the code on the work-stmadc-20191126 branch?

cd ~/klipper ; git fetch ; git checkout origin/work-stmadc-20191126 ; sudo service klipper stop ; make ; make flash ; sudo service klipper start

The resulting Klipper log file will need to be attached here from any tests (with this branch or any other branch).

Another thing to test would be disabling heater_bed in the config to see if it impacts the results of the extruder measurements.

-Kevin

Rubber1Duck, 2019-11-27:
Here are we go....
Branch work-stmadc-20191126
Roomtemperature 22°C
Steps at heating und cooling down are still there.
[image]
After round about 8 Minutes colling down both temperatures stall at 32,9 °C

[klippy.log](https://github.com/KevinOConnor/klipper/files/3895976/klippy.log)

to disable the bed heater, i have to try later, because i'm busy for the next hours

Peter

Magoo0876, 2019-11-27:
It's not possible to simply deaktivate the heater_bed in the config because you imediately get:
No option 'sensor_type' in section: 'heater_bed'
But as you see in my post 2 days ago, heating up the bed only gave the same effect.

-Robert

Rubber1Duck, 2019-11-27:
Robert,
it is possible, you have to delete the whole [heater_bed] section! # are not enough.
Roomtemperatur between 22 und 23 °C
Printer shows up 18,5 °C (19°C at the Display)
[image]

There is one Step at heating up und manny steps at cooling down. for more than 3 Minutes the Temperatur shows up no change !

[klippy.log](https://github.com/KevinOConnor/klipper/files/3898793/klippy.log)

Made the same check with Marlin, Same Roomtemperatur 22°C Marlin shows up 21 at the Display.
But there are the same Steps at heating und cooling down !?

I am very confused and completely at a loss!

KevinOConnor, 2019-11-28:
I suspect this is a board level issue, and not a software issue.  (Specifically, I suspect the thermistor circuit on the board isn't well designed and it is therefore not able to accurately measure voltages around 3V.  Thus resulting in inaccurate low temperature readings.)

Do you have the power jumper set to USB or 12V power?

-Kevin

Rubber1Duck, 2019-11-28:
Hi Kevin,

the SKR E3 DIP dont have this jumper!
I'm afraid you're right, it looks like a board level issue, but the temperature curves look better with marlin, even if they have these steps , but they are not so pronounced, and the Temperatur at Roomlevel differs only 1-2 °C. The result with klipper could be better!
Unfortunately, my programming skills with Python and C are more than bad, especially C is not my world which would be absolutely necessary for the firmware part.
To make sure that this possible board error only applies to my or possibly all SKR E3 DIP, I ordered another SKR E3 DIP yesterday, that arrives on Saturday and then I will test it again.
Because there are no further error messages in this direction.

Thanks for your great work and your support!
cu
Peter

KevinOConnor, 2019-11-30:
FWIW, in tests of my "skr mini v1.1" board, it also seems to show poor precision at low temperature.  I spent several hours on Wednesday trying to track it down.  My conclusion was that it was a hardware issue.  If Marlin is somehow "less broken" then I don't know why.

-Kevin

Rubber1Duck, 2019-12-01:
> 
> 
> FWIW, in tests of my "skr mini v1.1" board, it also seems to show poor precision at low temperature. I spent several hours on Wednesday trying to track it down. My conclusion was that it was a hardware issue. If Marlin is somehow "less broken" then I don't know why.
> 
> -Kevin

Hi Kevin,
if you have the same on your Board its rely looks like a board issue! So I have decided to send the new E3 DIP back and make no further tests.
For my opinion this issue will be closed.

cu
Peter

Magoo0876, 2019-12-02:
HI,
I have tried to solve the problem by adding some capacitors to the 3.3V. But nothing really worked. This is my fourth board now on my ender and hoped to stay with it a while.
From my side we can close here, if there is no genius idea anymore.

Thank's for your help.

Magoo0876, 2019-12-17:
I turned on my printer and could not trust my eyes. The temperature on the display showed 33° on he and 32 on the bed. Room was at 23°. 

[image]

So I updated Klipper and compiled the firmware, flashed the stm32, but still 32°
I thought my board gave up, but finally I tried to flash the original firmware.
[image]

[image]

So it is a firmware problem, maybe someone could look at the marlin 2.0 code to find their fix. 

--Robert

KevinOConnor, 2019-12-23:
Alas, someone with familiarity with the Marlin code will have to chime in.  (If anyone can even point to the *location* of the low-level ADC code that Marlin uses on stm32f103, that would help.)

-Kevin

Rubber1Duck, 2019-12-24:
i would say
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/HAL_STM32F1/HAL.cpp
Row 305 and following ....

KevinOConnor, 2019-12-24:
> i would say
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/HAL_STM32F1/HAL.cpp
Row 305 and following ....

That code invokes other code (like `adc.calibrate()`), but where does that code live?!

-Kevin

Rubber1Duck, 2019-12-24:
Hi Kevin,
i am realy not a expert in C and C++ ............. but for my understanding thats part of the framework-arduinoststm32-maple
and is inserted with #include <STM32ADC.h>
in https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/HAL_STM32F1/HAL.cpp
STM32ADC.h is located here
https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/STM32ADC/src

This library is pulled by Platformio witch is needed to compile Marlin 2.0 and up.

That is what i can provide with my no knowledge from C and C++ .....
maybe that helps
cu
Peter

Kirinkun, 2019-12-30:
Want to chime in to confirm that I've got a similar issue with my SKR Mini E3 on the Ender 3.

cmason1969, 2020-01-06:
Just adding another one with this issue. SKR mini E3 1.2 on an Ender 3. I am seeing temps around 5 degrees higher for room temperature readings.  Sorry I am too ignorant to add anything to the coding discussion but would help in testing if able.

Magoo0876, 2020-01-07:
I've  tried to survive with my own thermistor settings but the printer needs very long to start after heating and quits if you change the temperature during printing. Also the display shows strange hieroglyphs sometimes during printing like issue #2227.
In the meantime I forced a friend to run klipper on the same board, to see if it behaves different on his board. I also ordered a Cheertha 1.1b board with the same cpu to compare that to.

Hope to make progress, keep you informed.
 --Robert

Magoo0876, 2020-01-10:
In the meantime I could compare a known working boad with my toublemaker. What I've found you can see on the picture.  So maybe Kevin is right that it is a real hardware problem. So please check if you can also find spots like that shown in the picture.  Funny that the connecterd pins via the solder blob are also connected via the board, 

[image]

Stay tuned, Robert

Coffee0297, 2020-01-11:
we really soon need some fix for this. marlin works.... heres a pic of some with the mini E3. 
[image]

temps are not suposed to do this, and theres a lot of ppl that wanna get klipper and holds back bc this..

-Tonny

vladbabii, 2020-01-11:
The source of the image @Coffee0297 posted was me. This is on a skr mini 1.1

[image]

I have two temp sensors on a single bed
- blue one - analog pin 13 on mks gen 1.4
- red one - PA0 on skr mini 1.1

There is a 1500w, silicon heater under the bed. It's a 450x350x8 mm aluminium bed with glass on top. There's no way there's such a big temp fluctuation at 5 cm difference. 

[image]

[image]

Even my wife (which does not work/uses 3d printers and it's not technical) noticed somethings wrong by the smell of the hotend - "it smelled funny".

@KevinOConnor maybe we should add a warning about using these boards for extruders.

vladbabii, 2020-01-11:
During printing...
[image]
and zooming in...
[image]

vladbabii, 2020-01-12:
[image]

blue - 60.4
red - 52.3
target: 60
printer would keep trying to heat, wich would be dangerous if skr would not read the correct value soon - my bed probably can go above 150C

KevinOConnor, 2020-01-14:
FWIW, I'm aware of three differences between the Klipper adc code and the STM32ADC code (that Marlin apparently uses).  There is a slight difference in how the adc calibration is initiated, there is a slight difference in the analog sampling length, and the STM32ADC uses continuous sampling with DMA.  The first two are trivial and I'll make those changes to Klipper when I get a chance (unlikely for a couple of weeks).  None of these differences explain the poor accuracy however.

-Kevin

KevinOConnor, 2020-01-25:
FYI, I made the corresponding changes to sample time (bf3fa979) and calibration (502e8372).

If you do want to test this, grab the latest code (see the FAQ for details), rebuild the micro-controller code, flash it to the board, and make sure to fully power-off the board for 30 seconds after flashing (to ensure any new calibration takes effect).

I'm not expecting much difference, but if you do run a test let us know what the results are (success or failure).

-Kevin

Rubber1Duck, 2020-01-26:
I will test this a.s.a.p. and let you know what the results are .....

mental405, 2020-01-30:
Has anyone tried hooking one of these up to a scope to see if there is any noise on the analog reference voltage pin?

I had an MKS Gen L that exhibited similar behavior at higher temperature due to voltage instability on the 5V rail. After much searching someone had mentioned putting some capacitors across a 5V and ground pin corrected the issue. I later hooked the board up to a scope and confirmed the noise was there.

I have also heard around the water cooler than there are a number of the first revision Mini E3's which supposedly had crappy regulators on them, but that was supposedly fixed recently with the new ones.

Magoo0876, 2020-01-31:
I also tried to put additional caps to all power rails. With supply from raspberry only, no fans or other rf generators, see post from 2 Dec, no noticeable change on the temp measurement. But had no chance do set up a scope so far.

KevinOConnor, 2020-02-02:
Does the latest code not improve results?

FWIW, one thing you could try is connecting "analog_gnd" to "gnd" on the boards.  (That is, use a jumper wire to connect the "gnd" pin on the thermistor plug to a "gnd" pin elsewhere on the board, and then wire the thermistors between that ground and the sense pin on the thermistor plug.)  It's not clear to me that the analog circuits on the btt boards is correct.

-Kevin

brianrjones69, 2020-02-16:
I tried the latest.  Mine seems to be 4-5 degrees high.  But when I set the bed temp to 60 degrees it goes to 60 degrees, as measured by my fluke IR thermometer anyway.  It did the same with the previous version also.   I am more worried about the hot end temp though.  It seems that I have needed to lower my temps but I have no idea of how to measure it.  I haven't tried Marlin, but i also compared the code and didnt really see any difference, other than DMA'ing the adc.  I did increase the 'ticks' to 7 and it seemed more stable.  

I also connected the gnd together as you suggested but didnt see any change.  It wasn't the greatest test though because I connected them while it was already powered, but no change to the values.

KevinOConnor, 2020-02-18:
Okay thanks.  Probably want the ground connected prior to power up as the goal would be to have the "analog ground" truly grounded as part of the adc calibration routines.  That said, this is a "long shot" anyway, so not worth spending a lot of time on.

-Kevin

nitej, 2020-02-22:
I’m experiencing the same issue with an skr e3 dip v1.0. I’ve tried the standard thermister ports as well as using another analog pin and the normal gnd (not analog gnd) with an external circuit (the 4.7k pullup and a 10uF cap) but the temp offset and staircase like curves are still present.

Does the stock marlin firmware have the same “lumpy” temperature curve below 100C? Some have mentioned that the stock btt shipped firmware gave correct room temp readings but I’m curious if the strange heating/cooling are a firmware or hardware issue at the ADC level in the stm32.

Coffee0297, 2020-02-22:
i ahve talkd with InsanetyAutomation, and he says marlin oversamples
takes 16 readings and averages. that is fixing the problem at them. 
and looking at the diagrams of the E3 there is a 3.3v direct to the thermistor circuitry and not a filtered 3.3 as on the 1.3 1.4 and so on.
[image]
if the board shuld be done right one shuld have 3.3v from after the LV1 coil.. 

i hope somthing like an over sampler can be implemented together with a faster temp update as described in https://github.com/KevinOConnor/klipper/pull/2538#issuecomment-589521416
to make us able to get more presice temp readings and fix the E3 issue.

-Tonny

KevinOConnor, 2020-02-24:
This issue is not related to oversampling nor pid loop frequency.  Klipper implements 8x smoothing on each 300ms sample followed by 2s (or see config for `smooth_time`) smoothing on top of that.  Previous reports indicate the issue can also appears on Marlin (eg, https://github.com/KevinOConnor/klipper/issues/2194#issuecomment-559254285 ).  My understanding of the issue is that the adc sensor on these BTT boards quickly lose precision when trying to measure voltages above ~3V.  Thus resulting in "step transitions".

-Kevin

vladbabii, 2020-02-24:
@KevinOConnor so would using a logic level convertor to drop voltage from to 0-3 range ( or 0-2.8 ) might fix this? This would involve also adding a new mapping for the new range for the sensor, right ?

KevinOConnor, 2020-02-24:
@vladbabii - I don't know.

-Kevin

brianrjones69, 2020-02-25:
Kevin,
I did a little more looking into this issue.  It seems that the adc is a bit sensitive to source impedance and capacitance.  BTT probably should have put an op amp between the thermistor and the adc inputs.  That being said, it looks like the only way to get better data is to slow the conversion rate down, especially at higher voltages.  I found that using a ticks value of 6 brings the temp down, at least at lower temp levels.  It does not eliminate the steps, but its a more believable number at room temp.  The good news is that at lower voltages (higher temps), it seems that the adc is able convert the value much more accurately.  I assume it is because it can charge the sampling capacitor fully at the lower voltage.  At this point I have no choice but to believe that the issue is hardware related.  

Is there an easy way to get debug messages from the micro to the klipper.log?  I would like to see the adc values.

Brian

vladbabii, 2020-02-25:
How hard would it be to support this board (I2C) https://www.adafruit.com/product/1085 ?
This way we could have as many analog inputs as we want easily and cheaply.

Instead of trying to fix a broken product (and I'm sure it's not the only one), supporting a cheap analog board via SPI/I2C would solve the issue and also allow more versatility in klipper long-term

nitej, 2020-02-25:
Another approach (the one I am using) is to use a second mcu to handle the analog sensors and use the multi-mcu support built into klipper. The second mcu could be another printer board or anything else already supported (I’ve tried an extra arduino uno and a arduino Leonardo with an external resistor/cap circuit to form the voltage divider for the thermister). It should in theory just need to provide adc channels. The thermisters should be relatively slow signals compared to the stepper signals, so I don’t anticipate problems. Plus it’s already supported and breakout boards with a atmega32u4 are very inexpensive. 

A bit of a bandaid fix but if the skr e3 has a hardware fault, the alternative is using a different board altogether (or live with the wonky analog behavior).

KevinOConnor, 2020-02-25:
> BTT probably should have put an op amp between the thermistor and the adc inputs.

FWIW, an op amp isn't needed.  The simple voltage divider circuit works fine on many many boards.  As near as I can tell, BTT has done something dubious on these boards.  In particular, some of the schematics show a resistor between the adc pin and the divider which looks wrong to me (it increases impedance for no reason that I can see).  Some of the boards show an inductor between gnd and analog gnd which also looks odd to me.

> Is there an easy way to get debug messages from the micro to the klipper.log? I would like to see the adc values.

If you add output() calls to the micro-controller code, the messages should show up in the log.  See docs/Code_Overview.md for details.  You can also add `logging.info()` calls to the host code.

> How hard would it be to support this board (I2C)

As @nitej reports, using a low-cost off-the-shelf micro-controller is far simpler and cheaper.

-kevin

l-Kage-l, 2020-02-25:
Ok, so since i discovered this post i wanted to chime in with a solution of mine.

I tried updating Klipper and then re-flashing the new updated version, which broke Klipper temps. Marlin works well. Thing is, when i re-flashed the old one, the temps are working perfectly fine which is intriguing. Im going to post the ones i have working and the one that doesnt work for my SKR E3 Mini v1.3

WORKING:

[firmware.zip](https://github.com/KevinOConnor/klipper/files/4252524/firmware.zip)

Temps are reading fine, just like Marlin.

NOT working:

[klipper.zip](https://github.com/KevinOConnor/klipper/files/4252525/klipper.zip)

Temps are just as stated above, hovering around 32/33 degrees.

The version that doesnt work has the uint32_t aticks set to 6, as suggested in the comments above. I tried flashing a version with those set at 4 and at 6, and neither works. Hopefully this helps find the problem.

brianrjones69, 2020-02-25:
Yep.  Your working version is working for me too.  Thanks.

l-Kage-l, 2020-02-25:
No probs! Hopefully Kevin seems the message and figures what was the problem between versions.

KevinOConnor, 2020-02-26:
> I tried updating Klipper and then re-flashing the new updated version, which broke Klipper temps. 

What's the Klipper version that is working?  What is the Klipper version that is not working?  If you can produce logs of both then that would help.

Note that a single temperature check doesn't mean much because when a loss of precision occurs the "step temperature" might just randomly match the actual temperature.  We're looking for temperature changes to show a smooth transition without steps (or, at least, not such drastic steps).  So, a good test would be to try commanding the temperature up 25 degrees or so to see if the transition appears smooth.

If you, or someone else, has time, it would be extremely helpful to run a `git bisect` session to determine the Klipper revision that results in the regression.

-Kevin

tbe, 2020-02-28:
I have done the bisect.

Good news: I found the commit that introduced the change in the temperature readings. It is bd6c25c9f8c28831d146f9077a0a2aa636c6e037

Bad news: The temperatures before this commit are wrong also ( same for the firmware provided by @l-Kage-l . These readings are 16°C for me, at a constant room temperature of 21°C. And Marlin is reading 21°C also.

KevinOConnor, 2020-02-28:
Alas, looking for correct/incorrect temperatures will not yield anything useful.  One needs to look for "temperature steps" during heating/cooling.

If you can confirm that bd6c25c shows "temperature steps" and the previous commit (0b0e5a91) does not, then please attach the two log files from those revisions (that include a session showing a heat and/or cooling cycle).

Thanks,
-Kevin

brianrjones69, 2020-02-29:
I found it and it is where tbe said..  The problem is in stm32f1.c in clock_setup()  we are losing the prescalers in the rcc->cfgr registers.  I am surprised the thing works at all.
Here is my mod for the code.

Kevin, If you really need me to do a PR then I will, but its only 2 lines of code.

```
// Main clock setup called at chip startup
static void
clock_setup(void)
{
    // Configure and enable PLL
    uint32_t cfgr;
    if (!CONFIG_STM32_CLOCK_REF_INTERNAL) {
        // Configure 72Mhz PLL from external crystal (HSE)
        uint32_t div = CONFIG_CLOCK_FREQ / CONFIG_CLOCK_REF_FREQ;
        RCC->CR |= RCC_CR_HSEON;
        cfgr = (1 << RCC_CFGR_PLLSRC_Pos) | ((div - 2) << RCC_CFGR_PLLMULL_Pos);
    } else {
        // Configure 72Mhz PLL from internal 8Mhz oscillator (HSI)
        uint32_t div2 = (CONFIG_CLOCK_FREQ / 8000000) * 2;
        cfgr = ((0 << RCC_CFGR_PLLSRC_Pos)
                | ((div2 - 2) << RCC_CFGR_PLLMULL_Pos));
    }
// ------- here
    cfgr |= (RCC_CFGR_PPRE1_DIV2 | RCC_CFGR_PPRE2_DIV2    
                 | RCC_CFGR_ADCPRE_DIV4);
    RCC->CFGR = cfgr;
// ------- end here

    RCC->CR |= RCC_CR_PLLON;

    // Set flash latency
    FLASH->ACR = (2 << FLASH_ACR_LATENCY_Pos) | FLASH_ACR_PRFTBE;

    // Wait for PLL lock
    while (!(RCC->CR & `RCC_CR_PLLRDY))`
        ;

    // Switch system clock to PLL
    RCC->CFGR = cfgr | RCC_CFGR_SW_PLL;   // --------- here
    while ((RCC->CFGR & RCC_CFGR_SWS_Msk) != RCC_CFGR_SWS_PLL)
        ;
}
```

KevinOConnor, 2020-02-29:
Ouch!  That's a really bad error.  Thank you for tracking it down.  Commit 8dc431be.

-Kevin

brianrjones69, 2020-02-29:
You are welcome.  Thank you for providing all of us klipper!!

RaymondHimle, 2020-03-01:
This commit seems to have made the issue a fair amount less noticeable. Still a little bit 'steppy' but nothing like it was before this commit.

markwinger, 2020-03-04:
I just included this change in my klipper install.  I was seeing the huge stepping issue.  The change did seem to improve the stepping issue but the temp of the hotend reads low.  Hot end read 17C but room temp was about 25C.   Bed temp read 23C, closer but I would expect them to be the same. (Ender 3 with skr mini e3 v1.2).  I had and mks gen-l board before and the temps were smooth and at room temp the were the same and withing 1 or 2 degrees C.

mental405, 2020-03-05:
I would think that points to a variance with the thermistor or configuration or it since both the bed and the hotend should be using the same ADC.

tbe, 2020-03-05:
> I just included this change in my klipper install. I was seeing the huge stepping issue. The change did seem to improve the stepping issue but the temp of the hotend reads low. Hot end read 17C but [...]

Same here. Interestingly this is only an issue at room temperature. As soon as the bed or hotend are heating up, the temperature is very accurate

markwinger, 2020-03-05:
After leaving the system off over night the thermisters both read the same,17C and room temp is about 23C.  I think the bed must have actually been warmer.  So I read 6 degress low at room temp.  My bed typically reads about 5 or 6 degrees low when heated (using a non-contact infrared thermometer).  But I cant read the temp of the hotend with that so I don't know if the hot end is low at high temp but I expect the temps are both off by the same amount.

[111 more replies]

closed 2020-06-22 by KevinOConnor