Klipper3d/klipper #2196: 128x64 OLED IIC

iamacell, 2019-11-22:
Hello All!

I'm trying to connect an OLED screen to my printer as in the attached picture. 
No sign of life, the screen is blank (5V and GND are OK). What do I have to do to make the screen work? 
Thank you!

[image]

ShohninDmitriy, 2019-11-23:
[image]
[image]

KevinOConnor, 2019-11-23:
To the best of my knowledge the ssd1306 chips are 3.3v only, and it would not be safe to directly drive them from a 5v board.

-Kevin

iamacell, 2019-11-23:
@ShohninDmitriy thank you for the info! How can I set it in Klipper?
@KevinOConnor the data sheet of the screen said 3-5V. I've used it from 5V before (in an Arduino project) and worked fine. (Maybe it's not 1306?)

ShohninDmitriy, 2019-11-23:
> How can I set it in Klipper?

[display]
lcd_type: ssd1306
menu_timeout: 30
encoder_pins: ^P3.26, ^P3.25
click_pin: ^!P2.11

My settings for SKR1.3

ETE-Design, 2019-11-23:
@KevinOConnor Mine OLED also work on 5V, but how to tell Klipper to use the I2C as the display?
@ShohninDmitriy So you don't have to tell Klipper to use the I2C output, so if i don't have an encoder i just tell lcd_type: ssd1306?

ShohninDmitriy, 2019-11-23:
> So you don't have to tell Klipper to use the I2C output, so if i don't have an encoder i just tell lcd_type: ssd1306?

Yes

ETE-Design, 2019-11-23:
@ShohninDmitriy Mine just stays black, is that normal, or should it show some info also when it is not printing?

iamacell, 2019-11-23:
Same here. Blank/dark screen :(

ShohninDmitriy, 2019-11-24:
> @ShohninDmitriy Mine just stays black, is that normal, or should it show some info also when it is not printing?

everything works, maybe you connected incorrectly
[image]

[printer.zip](https://github.com/KevinOConnor/klipper/files/3883718/printer.zip)

ETE-Design, 2019-11-24:
@ShohninDmitriy  Funny mine still dosen't work either... Have a SKR Pro board... You don't have to crossover the connection right?

ShohninDmitriy, 2019-11-24:
> SDA-SDA, SCL-SCL, GND-GND, 5V-5V

That's right and connected.

ETE-Design, 2019-11-24:
@KevinOConnor Is the I2C part integrated in the SKR Pro yet, and do so need to add something to get it work?

closed 2019-11-29 by klipper-gitissuebot

conqueror433, 2021-06-12:
@ShohninDmitriy 

> > How can I set it in Klipper?
> 
> [display]
> lcd_type: ssd1306
> menu_timeout: 30
> encoder_pins: ^P3.26, ^P3.25
> click_pin: ^!P2.11
> 
> My settings for SKR1.3

I am trying to connect ssd1306 to SKR 1.3 for my Voron V0.  I connected SCL to Pin 1.23 & SDA to Pin 0.28 but screen is blank & when I add below setting to Klipper config, I am getting " MCU 'mcu' shutdown: i2c timeout" error.
[display]
lcd_type: ssd1306

Please share which Pins you connected SCL & SDA from SSD1306. Also suggest if I have to change settings in Klipper config.

ShohninDmitriy, 2021-06-13:
P0.0 sda P0.1 scl  - dir and step E1

conqueror433, 2021-06-13:
> P0.0 sda P0.1 scl - dir and step E1

Thanks a lot. It worked..!!