Klipper3d/klipper #2197: Missing "duration" in BLTouch.py

w1ebr, 2019-11-22:
Hello Kevin,
I have been having trouble with my new V3.1 BLtouch. In reading the bltouch.py code, I see at line 108 that the "reset" command does not include a "duration". When my BLtouch raised an alarm during probing, retrying did not help. After adding "duration=self.pin_move_time" to the reset command on line 108, the bltouch now resets and continues with the bed probe.
Gene

KevinOConnor, 2019-11-26:
A log from a case where this occurs (or really, a log in general from the printer) would help.  Without the low-level configuration details, there's little I can do.

-Kevin

w1ebr, 2019-11-26:
Kevin, I have added the pin-up duration to the "reset" command at line 108 in BLtouch. I think it was an oversight. The code now resets my BLtouch when it has a problem. I thought that was enough information for you.
Gene

closed 2019-11-29 by klipper-gitissuebot