Klipper3d/klipper #2198: Klipper crashes on last commit

mrvanes, 2019-11-22:
Just pull'ed klipper up to 00fd6223edcf0b80137779918c90621c18cf6a29 and am now experiencing crashes mid-print (unpredictable, but reproducable). As there were some changes in the firmware I also rebuilt and flashed the fw, so that may be related.
I rolled back to 9055e9f9b575efd3948651f2f11d1f671cef00ea and restarted the print (without reflashing) which succeeded succesfully.

Attached is the end of the klipper log, from the last succesful command to the crash.
[tail-klipper.log](https://github.com/KevinOConnor/klipper/files/3880739/tail-klipper.log)

KevinOConnor, 2019-11-22:
Can you attach the full Klipper log file from the incident?  Alas, without the full log there is little we can do.

-Kevin

mrvanes, 2019-11-23:
That felt like overkill, but lesson learned. Will checkout head again after this print and try to reproduce.

mrvanes, 2019-11-23:
Full klipper log of crashing print, using HEAD c188aa982f47a0e66e0907d1c39b5e20dcbc4af7
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3882517/klippy.zip)

KevinOConnor, 2019-11-24:
The stats messages suddenly stop at the end of that log, which makes me wonder if the host process got into an infinite loop.  Perhaps bdc7383 sneaked back in.

Does the issue occur in "batch mode" (as described in docs/Debugging.md)?  That is, does something like the following result in an infinite hang: `~/klippy-env/bin/python ~/klipper/klippy/klippy.py ~/printer.cfg  -d ~/klipper/out/klipper.dict  -i /path/to/troublesome.gcode -o /dev/null`

If so, does disabling pressure_advance in the printer.cfg file make the problem go away?  If so, does the following patch make the problem go away?
```diff
--- a/klippy/chelper/itersolve.c
+++ b/klippy/chelper/itersolve.c
@@ -160,7 +160,7 @@ itersolve_generate_steps(struct stepper_kinematics *sk, double flush_time)
         if (end > flush_time)
             end = flush_time;
         if (check_active(sk, m)) {
-            if (sk->scan_future && start > last_flush_time) {
+            if (sk->scan_future && start > last_flush_time + .000000001) {
                 // Must generate steps leading up to stepper activity
                 force_steps_time = start;
                 if (last_flush_time < start - sk->scan_future)
```

-Kevin

mrvanes, 2019-11-24:
> Does the issue occur in "batch mode"

Yes

> If so, does disabling pressure_advance in the printer.cfg file make the problem go away?

Yes

> If so, does the following patch make the problem go away?

Yes

I'm starting to feel sorry for running klipper on ancient, crappy, buggy x86 fp hw ;)

KevinOConnor, 2019-11-24:
The hardware is fine - it's just that gcc compiles for 387 chips by default on 32bit.  I suspect if you add "-msse2 -mfpmath=sse" to COMPILE_CMD at the top of `klippy/chelper/__init__.py` and forced a rebuild (`rm ~/klipper/klippy/chelper/c_helper.so`) then the problem would also go away.

Alas, I fear the fix I sighted above isn't numerically stable, so I'll have to think of a more permanent fix.

Thanks for reporting.
-Kevin

mrvanes, 2019-11-28:
Building using -mpc64 gcc flag also at least one time succesfully completed the forementioned debug run without the offset patch and has no visible impact on print quality.

https://stackoverflow.com/questions/503436/how-to-deal-with-excess-precision-in-floating-point-computations

KevinOConnor, 2019-12-10:
Thinking about this further, I think the code change tested above should be okay even in the general case.  I went ahead and committed it (4cbbe18a).

-Kevin

KevinOConnor, 2019-12-23:
I'm going to close this as I believe the issue is now fixed.

-Kevin

closed 2019-12-23 by KevinOConnor