Klipper3d/klipper #2199: incremental/adaptive bed autolevelling

antst, 2019-11-23:
I was thinking, why we need to probe whole bed? 
Why to not have something which probes bed only in printing area before the print? 
And those points can be stored, to accumulate.
If bed is very large and necessary very flat (as likely to be the case with scara, where people sometimes print on desks, or even scara with full 360 degrees rotation ), this might be quite good approach.

JohnEdwa, 2019-11-23:
Marlin ABL lets you do exactly this using G29 with the BFLR (back/front/left/right) and X/Y (grid size) parameters to change the probing points at runtime when issuing the command. Klipper currently doesn't, because the parameters have to be hard-coded to the config files and the probe commands don't calculate offsets by themselves (yet, pullrequest #1949 aims to solve that)
  
To automate it, the slicer you use would need to provide the model boundaries as variables, do you know which ones do that?

antst, 2019-11-23:
I was thinking about writing post-processing script which estimates area of the first layer from g-code. I am unaware of slicers doing that. Though, with open-source slicers it must be possible to add this easily. After all, they already have code for skirt, which does exactly that.

KevinOConnor, 2019-12-01:
I'm not really sure what's being proposed here.  As far as I know, one can already alter the bed mesh calibration region in the config file so that it only probes the given region.

-Kevin

JohnEdwa, 2019-12-01:
But you cannot do that from a gcode file, and not without requiring Klipper to be rebooted. Unless there is some "modify configuration parameters and refresh Klipper" procedure I've missed. 

The idea proposed is that instead of always probing the whole bed at the start of a print, just the actual first layer area of the model about to be printed is probed, saving a lot of time. To be able to do that, there would need to be a way to specify the area to be probed as a variable using a Gcode parameter.

For Marlin, a command that does it that would look something like `G29 B150 F50 L100 R200 P3`.

KevinOConnor, 2019-12-13:
Okay, I understand.  Ultimately, it's up to @arksine and the developer implementing it, but it sounds like a reasonable feature to me.

-Kevin

KevinOConnor, 2020-01-03:
I'm going to close this as it looks like the conversation has concluded.  Let us know if someone decided to take on implementation of this.

-Kevin

closed 2020-01-03 by KevinOConnor