Klipper3d/klipper #2201: Unknown value 'XDIAG1' in enumeration 'pin'

terrorhai, 2019-11-23:
Hey ho,

trying to setup sensorless homing in my Kossel Linear Plus.
HW is SKR 1.3 and Bigtreetech TMC2209 V1.2.

As you can see, I always get an error **(Unknown value 'XDIAG1' in enumeration 'pin')** when trying to reload the config.
So, what am I doing wrong?

[tmc2209 stepper_a]
uart_pin: P1.17
microsteps: 16
run_current: 0.750
hold_current: 0.750
stealthchop_threshold: 250
driver_SGTHRS: 250
diag_pin: XDIAG1

[stepper_a]
step_pin: P2.2
dir_pin: !P2.6
enable_pin: !P2.1
step_distance: .0125
endstop_pin: tmc2209_stepper_a:virtual_endstop
homing_speed: 5
# The next parameter needs to be adjusted for
# your printer. You may want to start with 280
# and meassure the distance from nozzle to bed.
# This value then needs to be added.
position_endstop: 295.6
arm_length: 271.50

BlackStump, 2019-11-23:
the pin is not correct?
ie https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-v1.3.cfg#L140

terrorhai, 2019-11-24:
> 
> 
> the pin is not correct?
> ie https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-v1.3.cfg#L140

I tried it out, but I got TMC2209 which are using UART, not SPI.

KevinOConnor, 2019-11-25:
XDIAG1 is not a pin on the micro-controller.  Your micro-controller has pins named like "P1.29".  You'll need to find the micro-controller name of the pin you wish to use.  The example config file for the SKR 1.3 does have the micro-controller pin names so you should be using that as a reference.

-Kevin

glowtape, 2019-11-26:
Take a look at the section on the left marked TMC2130-SPI in this file:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/hardware/SKR-V1.3-PIN.pdf

These also applies to sensorless homing via UART. You need to put a jumper on XDIAG1/X-. This'll connect the DIAG pin of the X-axis driver to the X- endstop, which is P1.29 per pin definitions a bit lower on the schema. Same applies for the other axes.

terrorhai, 2019-11-26:
Wohoo, it works!
Sorry for beeing dumb and not double checking the wiring diagramm. The german manual for the SKR says, to disconnect the jumpers when using UART.

glowtape, 2019-11-26:
Yeah, things are a bit unclear about all that.

closed 2019-11-27 by terrorhai