Klipper3d/klipper #2202: Force move causes shutdown Timer too close

doowrednu, 2019-11-24:
Hi Kevin - Just updated to latest version of Klipper and flashed both Arduinos (Ramps)

When I try to issue a force move command 
```FORCE_MOVE STEPPER=stepper_z3 DISTANCE=0.05 VELOCITY=1```

I get a shutdown with Timer too close error. 

To rule out this was not specific to this printer / comms etc. I just re-flashed a second printer and get exactly the same behaviour. I can home, quad, and print no problem - the force move always produces the problem. 

```
Read 6346473.269765: 'FORCE_MOVE STEPPER=stepper_z3 DISTANCE=0.05 VELOCITY=1\n'
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[0.0, 0.0, 0.0, 0.0] homing_position=[
0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=25.0
MCU 'mcu' shutdown: Command request
clocksync state: mcu_freq=16000000 last_clock=107361805 clock_est=(6346467.352 18834906 15878454.803) min_half_rtt=0.001559 min_rtt_time=63464
72.926 time_avg=6346467.350(2.859) clock_avg=18834906.346(45400877.944) pred_variance=154693209.550
Dumping serial stats: bytes_write=887 bytes_read=3830 bytes_retransmit=0 bytes_invalid=0 send_seq=106 receive_seq=106 retransmit_seq=0 srtt=0.
004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0
```

Log attached - Thanks
[klippy.log](https://github.com/KevinOConnor/klipper/files/3883201/klippy.log)

alecplumb, 2019-11-24:
I see the same issue with dual ramps.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3883232/klippy.log)

KevinOConnor, 2019-11-24:
Thanks.  I think this should be fixed now (commit ade65b90).

-Kevin

doowrednu, 2019-11-24:
Hi Kevin - Thank you - yes that fixes it for me

closed 2019-11-24 by doowrednu