Klipper3d/klipper #2203: Different behavior  if macros is executed via vSD or Octoprint stream them

Hywelmartin, 2019-11-24:
I have a start script that is a bit over the top, maybe.... [video](https://youtu.be/AoZJLQzpFMY)
 
And i have a pause i the middle of the routine where i load filament.. 

The slicer creates this:
```
; generated by PrusaSlicer 2.1.0+linux64 on 2019-11-23 at 18:52:18 UTC

; /home/nitram/appimager/M204.40.pl;
; 
; till temptower +5⁰C  vid "varje färgbyte"
; 
; /home/nitram/appimager/M600_TO_NISSE.pl;

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.47mm

M107 ; disable fan
THE_START B=50 H=230
M140 S50  ; set bed temperature 
M104 S230 ; set temperature
THE_START_END
;SET_VELOCITY_LIMIT ACCEL=7000 ACCEL_TO_DECEL=1500 SQUARE_CORNER_VELOCITY=20
SET_VELOCITY_LIMIT ACCEL=7000 ACCEL_TO_DECEL=1500 SQUARE_CORNER_VELOCITY=12
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
; Filament gcode PA was 0,25
;SET_PRESSURE_ADVANCE ADVANCE=0.14 ADVANCE_LOOKAHEAD_TIME=0.01
SET_PRESSURE_ADVANCE ADVANCE=0.24  SMOOTH_TIME=0.015
;nu byter jag till lager 0.2
;;
;
;
G1 E-1.20000 F1800.00000 ; retract
G1 Z0.200 F30000.000 ; lift Z
;nu har jag bytt till lager 0.2
;;
M117 LAYER  #0 @ 0.2mm
G1 Z0.400 ; lift Z
G1 X19.830 Y-14.174 ; move to first skirt point
G1 Z0.200 ; restore layer Z
```
I jumped on the smoothed pressure advance pull request and reverted back to master yesterday..
so in that time frame something happened...

[klippy191123.log](https://github.com/KevinOConnor/klipper/files/3883572/klippy191123.log)

Arksine, 2019-11-24:
@KevinOConnor We appear to have a deadlock in the virtual_sdcard module due to the recent change to wait until the work_timer is set to `None` before returning from do_pause().   Unfortunately that never happens if the PAUSE originates from the work timer itself, because the work timer is waiting for run_gcode() to return.

We didn't catch this when you made the initial change because we were always sending the PAUSE from Octoprint.  We might need a mechanism to track the source of the pause, if its external we wait, if its internal we need to return immediately from do_pause().

KevinOConnor, 2019-11-24:
@Arksine - Good catch.  Should be fixed now (commit 05472bb8).

-Kevin

Hywelmartin, 2019-11-24:
works...

Thanks for super-fast response 

closing

closed 2019-11-24 by Hywelmartin