Klipper3d/klipper #2204: Connection Error -> 'pressure_advance_lookahead_time'

TheRunnerPi, 2019-11-24:
Hi,

i can´t Connect to Klipper. The Printer.cfg ist copied from another Printer. I did change the USB-Port.

Error:   
Option 'pressure_advance_lookahead_time' is not valid in section 'extruder'

[klippy.log](https://github.com/KevinOConnor/klipper/files/3883662/klippy.log)

Hywelmartin, 2019-11-24:
as the error say 
Due to new  PA code there is a new variable and the old one is gone..
have a look at [[extruder] ](https://github.com/KevinOConnor/klipper/blob/c188aa982f47a0e66e0907d1c39b5e20dcbc4af7/config/example.cfg#L96) for valid settings

TheRunnerPi, 2019-11-24:
Thanks to @Hywelmartin 

I changed the **pressure_advance_lookahead_time** to **pressure_advance_smooth_time**

Now the Connection Error is **Printer is not ready**

[klippy.log](https://github.com/KevinOConnor/klipper/files/3884052/klippy.log)

TheRunnerPi, 2019-11-24:
`serial = /dev/serial/by-id/usb- usb-Arduino__www.arduino.cc__0042_5583832373535161E050-if00`
was wrong

closed

closed 2019-11-24 by TheRunnerPi