Klipper3d/klipper #2212: Multi MCU shutdown during commissioning

SrGeek, 2019-11-25:
I am running a Re-Arm and Mega2560 board.  Re-Arm-x,x1,y,E0,E1,T0,T1,estops x,x1,y. heaters E0/E1.  Mega2560-z0,z1,bed heater,endstop z (touchmi).
During commissioning I send the following command STEPPER_BUZZ STEPPER=stepper_z.  And got this message:
Recv: // Klipper state: Shutdown

then I issued STEPPER_BUZZ STEPPER=stepper_x and got the following message:

Recv: // MCU 'zboard' shutdown: No next step
Recv: // This is generally indicative of an intermittent
Recv: // communication failure between micro-controller and host.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'zboard' shutdown: No next step

Please advise how I can correct this.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3886529/klippy.log)

Hywelmartin, 2019-11-25:
#2202 maybe

SrGeek, 2019-11-26:
I uploaded latest commit.  flashed my Re-Arm and Mega without success.  See attached klippy.log file.  It definitely happens when I Stepper_Buzz.  Maybe there is a timing correction that needs to occur between both boards?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3889936/klippy.log)

KevinOConnor, 2019-11-26:
Alas, you've hit a quirk between stepper_buzz and square_corner_velocity.  stepper_buzz doesn't do acceleration, and the lack of accel/decel is tickling the safety check configured in to the micro-controller.  Simplest solution is to not set square_corner_velocity=1 during commissioning.

-Kevin

SrGeek, 2019-11-29:
Re-Arm not crashing now.  Remove the square velocity from the printer kinematics section.

closed 2019-11-29 by SrGeek