Klipper3d/klipper #2213: BL-Touch original - z_offset problem

Matrixer007, 2019-11-25:
Hi!
I installed the original V 3.1 without problems on the MKS Gen 1.4 under Klipper. However, the z_offset does not work as it should. It does not matter what value I enter, the nozzle is always 0.5mm above the bed. Anyone an idea where the problem lies?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3887841/klippy.log)

Hywelmartin, 2019-11-25:
I don't know if it helps, but your z_offset is negative.... I believe that it should be the actual distance between the nozzle and the bed when the probe triggers

Matrixer007, 2019-11-25:
It does not matter what value I enter in the job, the result is always the same, that's the problem!

BlackStump, 2019-11-26:
suggestion check you do not have a mechanical issue ie a z endstop still in place that prevents movement below 0.5mm
Bltouch usually have a positive offset of around 1.5-2mm so it does matter, a negative number is saying your nozzle will be below the bed surface when your probe triggers.

Matrixer007, 2019-11-26:
No, there is no mechanical issue in place, I also use the Touch as Z-Endstop.

KevinOConnor, 2019-11-26:
I recommend you follow the guide at https://www.klipper3d.org/Probe_Calibrate.html to calculate and save the probe's z_offset.  If you don't have success with that, please attach the klipper log file from the session that contains the PROBE_CALIBRATE and ultimate SAVE_CONFIG commands.

-Kevin

darkernoise, 2019-12-02:
@Matrixer007: Your BLTouch Z offset is negative: z_offset = -3.0.  It should be positive.  If it helps, the z_offset is the distance between the Z value where the probe triggers and the positive additional distance to the tip of the nozzle.

Matrixer007, 2019-12-02:
Sorry, I am too late. The Calibration like described from Kevin was successfull!
Thanks for help!

closed 2019-12-02 by Matrixer007