Klipper3d/klipper #2214: STM32F446 / Fysetc S6 Can't connect

towe96, 2019-11-25:
I flashed my Fysetc S6 with the klipper.bin file, but my Raspberry Pi 4 can't seem to detect it when plugging it in. It doesn't show up under "lsusb".

I made no changes in the "make menuconfig" settings except for the STM32F446 MCU.
I flashed the board using STM32CubeProgrammer to start address 0x08000000.

Any clues? Does anyone have Klipper working on a F446 board?

KevinOConnor, 2019-11-25:
At a minimum that board has a 12Mhz crystal.  You can pull the latest Klipper code, select "low level options" in "make menuconfig", and try with the new 12Mhz clock selection.

-Kevin

towe96, 2019-11-25:
Gorgeous, that seems to have been it.
I'll get my config ready tomorrow and upload it.
You've also got a new Patreon now - thanks for all the amazing work.

towe96, 2019-11-25:
Works perfectly fine now.

Still have some issues when using both 24V and 5V over USB at the same time, but that's board / janky PSU-related.

closed 2019-11-25 by towe96