Klipper3d/klipper #2215: SKR Mini E3 V1.0 - BLTouch - Always Triggered State

dshag86, 2019-11-25:
Unknown BLTouch version, maybe v2.0? 

No matter what configuration I use, or pin on the board, the BLTouch is always in a triggered state. I have tried both the pin_up_touch_mode_reports_triggered: True/False && pin_up_reports_not_triggered: True/False. I can successfully send pin up and pin down without issue. 

Attached is the log after doing a test-bltouch.gcode

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3888587/klippy.zip)

ld3300, 2019-11-29:
When you send your debug commands to the BLTouch does the LED go out when it is extended?

dshag86, 2019-11-29:
That's correct. Visually, the BLTouch works as intended.

ld3300, 2019-11-29:
I'm not very good with parsing the log, so I didn't see anything that stuck out to me. You can try setting up some feedback to see what the pin state is doing.
Add something like this to your config file:

```
[gcode_button tempSense]
pin: ^sensor_pin   #put sensor pin for BLtouch here, assign an unused pin to BLtouch temporarily
press_gcode:
    RESPOND TYPE=echo MSG="PIN UP"
release_gcode:
    RESPOND TYPE=echo MSG="PIN DOWN"
```
Then send BLTOUCH_DEBUG COMMAND=pin_down and see what the terminal output says
Also try with pin_up.

If neither of those produce a result you may want to try turning the adjustment screw on the top as the probe may not be retracting far enough to be sensed.

You can also try applying power and issuing the commands with a multimeter connected to measure the voltage on the sense pin

dshag86, 2019-11-29:
I'll try this and report back in a few days when I get a chance to tinker.

Thank you

dshag86, 2019-12-09:
Unknown command:"RESPOND"

Tried: RESPOND_TYPE and RESPONDTYPE

all three are unknown commands

KevinOConnor, 2019-12-13:
If you are able to control the bltouch, but are not able to read a signal from the bltouch, it generally indicates that the sensor pin is not configured with the correct pin or is not wired correctly.  The best advice we have is the instructions in at: https://www.klipper3d.org/BLTouch.html

-Kevin

dshag86, 2019-12-13:
So I have it wired correctly, however the code given to me above I believe
is incorrect as it returns an error for that line.

The issue is that klipper only sees the bltouch as triggered regardless of
state or pin configuration. So either my bltouch is all of a sudden broken,
or I’m missing something here.

KevinOConnor, 2020-01-03:
Alas, it doesn't seem anyone has any further advice to share here.  The best info I have is at: https://www.klipper3d.org/BLTouch.html

-Kevin

BlackStump, 2020-01-04:
just my personal annoyance if a log is zipped I will not look, creates extra steps and cleanup on my pc
but looking at your log you have the wrong sensor pin

[bltouch]
sensor_pin: PC14
control_pin: PA1

BlackStump, 2020-01-04:
[image]

dshag86, 2020-01-04:
Without having the ability to look at the moment, are you suggesting this
is the correct pin config, or this is what I have and its wrong? I have
tried multiple pin configurations and mappings

Thanks!

BlackStump, 2020-01-04:
I have given you the correct pins and shown the connections on the board which are servo and probe.

dshag86, 2020-01-04:
Thank you, I believe I had the pin configuration correct per the schematic.
But I will try again with your suggestion when I have a chance to tinker.

Thank you again!

FanDjango, 2020-01-04:
Hmm, in your log I can see:
```
[bltouch]
sensor_pin = ^PC2
control_pin = PA1
x_offset = -45
y_offset = -10
z_offset = 1.6
pin_move_time = 0.4
```

In the diagram posted by @BlackStump I see it should be PC14, that's what he was saying.

So there is a discrepancy you need to clear up first I think.

dshag86, 2020-01-04:
The PC2 pin is the z end stop, which was suggested to be used instead of
PC14. However I do remember switching between the two when I was diagnosing
this issue a while back, still getting the same result.

FanDjango, 2020-01-04:
OkiDoki. What about connecting some real switch on that pin (you might have a spare one) INSTEAD of the BLTouch and testing with that?

By the way, the wires to connect for the sensor signal pin, some people wrongly believe that it is not important which one goes to ground and which one goes to the PC14. It is not like a real switch. If you put the black one into PC14 it will be continuously triggered because BLACK is GND and will pull down the PC14 pin. Make sure you have the WHITE one on the PC14.

dshag86, 2020-01-04:
I’ll check the wiring when I get a chance.
This is good insight, thank you.

KevinOConnor, 2020-02-02:
Okay, I'm going to close this as it looks like the conversation has concluded.

-Kevin

closed 2020-02-02 by KevinOConnor