Klipper3d/klipper #2219: Option 'pressure_advance_lookahead_time' is not valid in section 'extruder'

sinicolau6, 2019-11-26:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3889970/klippy.log)

I tried setting up pressure advance, i printed the Tuning Tower, got the value right, then when i tried writing the 2 lines of code under [extruder] this error shows up:

Option 'pressure_advance_lookahead_time' is not valid in section 'extruder'

What can i do?

[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
step_distance: .0125
endstop_pin: ^PC0
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: PB15
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
step_distance: .0125
endstop_pin: ^PC1
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_y]
uart_pin: PC6
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0.0
position_max: 250
position_min: -1.5

[tmc2209 stepper_z]
uart_pin: PC10
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 5

[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD2
step_distance: 0.01039965
nozzle_diameter: 0.400
filament_diameter: 1.750
pressure_advance: 0.1758
pressure_advance_lookahead_time: 0.010
heater_pin: PC8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
#control: pid
#pid_Kp: 29.16
#pid_Ki: 2.93
#pid_Kd: 72.47
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: PC11
microsteps: 16
run_current: 0.650
hold_current: 0.500
stealthchop_threshold: 5

[heater_bed]
heater_pin: PC9
sensor_type: ATC Semitec 104GT-2
sensor_pin: PC3
#control: pid
#pid_Kp: 54.027
#pid_Ki: 0.770
#pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PA8

[mcu]
serial: /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[static_digital_output usb_pullup_enable]
pins: !PC13

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PB6, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB7, EXP1_10=<5V>

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2

[output_pin beeper]
pin: EXP1_1

[bltouch]
control_pin:PA1
sensor_pin: PC2
x_offset: -42.6
y_offset: -16.4
#z_offset: 0
pin_move_time: 0.4

[bed_mesh]
speed: 100
horizontal_move_z: 10
min_point: 90, 30
max_point: 235,235
probe_count: 3, 3
fade_start: 1.0
mesh_pps: 2,2
relative_reference_index: 4

[gcode_macro G29]
gcode:
	G28
	G1 Z10 F600
	BED_MESH_CALIBRATE
	BED_MESH_OUTPUT

[gcode_macro START_PRINT]
gcode:
  M204 P500.00 R1000.00 T500.00 
  M220 S100 
  M221 S100 
  G29
  G92 E0 
  G1 Z10 F3000 
  G1 X10.1 Y20 Z0.28 F5000.0 
  G1 X10.1 Y200.0 Z0.28 F1500.0 E15 
  G1 X10.4 Y200.0 Z0.28 F5000.0 
  G1 X10.4 Y20 Z0.28 F1500.0 E30 
  G92 E0 
  G1 Z2.0 F3000

[safe_z_home]
home_xy_position: 155,120
speed: 80.0
z_hop: 10.0
z_hop_speed: 10.0

   	

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# points =
#*# 	0.135000, 0.180000, 0.240000
#*# 	-0.002500, -0.015000, -0.045000
#*# 	0.135000, 0.092500, 0.052500
#*# x_count = 3
#*# y_count = 3
#*# min_x = 90.0
#*# max_x = 235.0
#*# min_y = 30.0
#*# max_y = 235.0
#*# x_offset = -42.6
#*# y_offset = -16.4
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 28.858
#*# pid_ki = 1.924
#*# pid_kd = 108.217
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 59.962
#*# pid_ki = 1.285
#*# pid_kd = 699.312
#*#
#*# [bltouch]
#*# z_offset = 3.980

BlackStump, 2019-11-26:
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md

fl1ck3r, 2019-11-26:
for the lazy, relevant section in example.cfg
`#pressure_advance: 0.0
#   The amount of raw filament to push into the extruder during
#   extruder acceleration. An equal amount of filament is retracted
#   during deceleration. It is measured in millimeters per
#   millimeter/second. The default is 0, which disables pressure
#   advance.
#pressure_advance_smooth_time: 0.040
#   A time range (in seconds) to use when calculating the average
#   extruder velocity for pressure advance. A larger value results in
#   smoother extruder movements. This parameter may not exceed 200ms.
#   This setting only applies if pressure_advance is non-zero. The
#   default is 0.040 (40 milliseconds).`

sinicolau6, 2019-11-26:
Thanks, i didnt know about smooth time change!

closed 2019-11-26 by sinicolau6