Klipper3d/klipper #2221: M567 Tool mix GCode

DMCShep, 2019-11-26:
M567 is currently not supported. In RepRapFirmware, this provides tool mixing ratios for single-out multi-in hot ends to allow a range of colors. See here:

https://duet3d.dozuki.com/Wiki/Gcode#Section_M567_Set_tool_mix_ratios

A quick search for `M567` in this repo provided exactly zero results -- code, issues, pull requests, or otherwise.

KevinOConnor, 2019-11-26:
Looks like a dup of #950 .

-Kevin

DMCShep, 2019-11-26:
Not quite, M163 is for setting the weight ratio, M567 sets percentage ratios without regard to weight. See here:

[G-Codes not implemented which are not planned for implementation](https://duet3d.dozuki.com/Wiki/G-Codes_not_implemented#Section_G_Codes_not_implemented_which_are_not_planned_for_implementation)
> M163: Set weight of mixed material ( use M567 to define the ratios based on extrusion percentage not weight)

KevinOConnor, 2019-12-01:
It's close enough to #950 that I think tracking there is preferable.

-Kevin

closed 2019-12-01 by DMCShep