Klipper3d/klipper #2222: Timer Too Close: Help interpreting the log file, please

drelich, 2019-11-27:
Hi,

I'm getting 'Timer too close' error a lot lately (even if printing from virtual SD card), printing fairly simple things. 

I found the info on how to do debugging, ran the py script, which returned a few log files, including the 'klippy.log.shutdown238652' one (attached but you'll need to remove the .log from the end of the file name - it wouldn't let me uploaded without it). How do I read them though? What am I looking for? They're incredibly long and complex for an untrained eye.

Thanks.

David.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3896545/klippy.log)
[klippy.log.shutdown238652.log](https://github.com/KevinOConnor/klipper/files/3896555/klippy.log.shutdown238652.log)

KevinOConnor, 2019-11-30:
A "timer too close" error indicates that the micro-controller got a request from the host to take an action at a time that has already past.

Reading the log requires detailed knowledge of the micro-controller commands and what they do (there's some detail in docs/MCU_Commands.md).  The shutdown log provides information on what requests were read and what commands were send around the time of the failure.  Specifically, in your failure:

```
Receive: 96 71390.195344 71389.990615 14: seq: 1c(42107), analog_in_state oid=20 next_clock=1369294336(71390.483957) value=609
...
Sent 95 71390.500004 71390.498164 46: seq: 1c(42108), schedule_soft_pwm_out oid=21 clock=1369422336(71390.491957) on_ticks=746368, ...
...
Receive: 99 71390.504360 71390.500004 12: seq: 1d(42108), shutdown clock=1369554986(71390.500247) static_string_id=Timer too close
```

The host got a temperature measurement and then responded to it ~305ms later, which is way too late.

It looks like you're running old code from @dmbutyugin's s-curve branch.  You should probably be either on the latest version of that test code or on the master branch.

-Kevin

drelich, 2019-12-02:
Thanks, @KevinOConnor!

I did a clean install of OctoPrint and Klipper, and printing off the virtual SD card now. So far so good.

Regarding the 305ms being way too long for a temp measurement response... If it's just a temp reading, is it really a reason to abort the print? Watching the terminal, temp measurements are so frequent, surely there's a way to abort only if a set number of measurement responses get delayed (which could be indicative of a serious problem with the thermistor, etc.)?

I am in no way an expert in MCUs and Klipper, I just feel like one slight delay not related to motion shouldn't be a cause for wasting half a roll of filament. 

Maybe make the print pause until it all catches breath and then resume. In most cases (according to my own observation) those aborts happened when the nozzle was above an infill area so there wouldn't have been any visible damage done to the print.

Cheers.

David.

KevinOConnor, 2019-12-02:
A "temp measurement response" is a command to set the heater power.  That is, the micro-controller got a request to update the heater pin at a time in the past.  The micro-controller doesn't know anything about the pin or its function, it only knows it got commanded to do something it could not do.

A normal heater update occurs in 5-10ms.  A delay of 305ms is indicative of a severe systemic problem.  FWIW, I want my printer to stop when it detects systemic issues so that I can fix the problem and then get high quality prints.  I do not want my printer to continue, as I fear that could lead to dozens of hours wasted trying to figure out why the printer is producing bad quality prints (or worse, never finding out why).

-Kevin

drelich, 2019-12-02:
Ah, okay. To me a "temp measurement response" means that "something" is asking to be updated on current temperature and is expecting a response of sorts. Didn't know it's actually a request to set a temperature, in which case I agree that it could result in bad print and so the print should be aborted. D.

closed 2019-12-02 by drelich

dmbutyugin, 2019-12-02:
FWIW, if I read the attached logs correctly, you've been on some of the older branches, which has advanced lookahead planning still implemented in Python. That code has known performance issues, e.g. it can run a single flush operation for a few seconds, way above a reasonable delay. It can lead to all sorts of timer issues. That's why I re-implemented that lookahead algorithm in C, so does not have such performance issues. It is also synced to the latest master branch, so all timer improvements should be in. If you want to continue using bezier acceleration, consider using scurve-c-combine-smoothpa branch instead of the old one.

drelich, 2019-12-02:
I'll try to install the latest version and see how it goes. Great stuff about re-coding it in C! :) Thanks!