Klipper3d/klipper #2223: /tmp/printer error

edanur95, 2019-11-28:
Hi,
i am using buster software with raspberry pi 4 and smoothieboard. After installing klipper I connect with / tmp / printer. But it doesn't read my temperature. The log file has an attachment. printer is not ready error. can you help me ?

[octoprint (10).log](https://github.com/KevinOConnor/klipper/files/3900653/octoprint.10.log)

edanur95, 2019-11-28:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3900992/klippy.log)

BlackStump, 2019-11-28:
always a good idea to check the change log here
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md
you need to follow this to find the new serial id
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

edanur95, 2019-11-28:
I tried that way. but ı can't control the smoothieboard from octoklipper.

Arksine, 2019-11-28:
You need to change the `serial` option under the `[mcu]` section in printer.cfg.  If you recently flashed then the device id has changed.  You can find it by connecting to your Pi via ssh and entering the following command:

```
ls /dev/serial/by-id/
```

KevinOConnor, 2019-12-13:
I'm going to close this as it looks like the original question was answered (be sure to read the Config Changes document on upgrades as described at https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software and re-run the `ls /dev/serial/by-id/*` command and update the printer.cfg file as described in the installation document at https://www.klipper3d.org/Installation.html ).

-Kevin

closed 2019-12-13 by KevinOConnor