Klipper3d/klipper #2224: Unable to connect Klipper from Octoprint, on Creality Ender 5

srekan, 2019-11-28:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3902119/klippy.log)
Attached is the log file

Arksine, 2019-11-28:
Try reflashing your MCU.  It appears to be connecting but unable to communicate.

srekan, 2019-12-03:
I tried it multiple times.
I followed this doc: https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

The below step was always failing
make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

Then I tried the below command which gave a success result. I found this in the comments section of https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender5-2019.cfg

avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex

Still facing the same issue.

Arksine, 2019-12-03:
Go into your ~/klipper directory and enter the following command:
 ```
cat .config
```
Copy and paste the output here.

Magoo0876, 2019-12-03:
Often the issue is to not enter the correct device id in the config or the make flash command.
So please check if your ID matches.
Open a teminal on Linux or PuTTY on Windows and connect via SSH. Enter the user and password and the following:

cd /dev/serial/by-id/
ls

Now you should see your ID which has to be used for the steps you allready did. Add the path and the ID und you are fine.

My output looks like that:
pi@octopi:~/klipper $ cd /dev/serial/by-id/
pi@octopi:**/dev/serial/by-id** $ ls
**usb-Klipper_stm32f103xe_02081976-if00**
pi@octopi:/dev/serial/by-id $ 

So for me it would be in theory: 
make flash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper_stm32f103xe_02081976-if0

-Robert

srekan, 2019-12-05:
@Arksine, attaching the file.
[myconfg.txt](https://github.com/KevinOConnor/klipper/files/3925050/myconfg.txt)

srekan, 2019-12-07:
> Often the issue is to not enter the correct device id in the config or the make flash command.
> So please check if your ID matches.
> Open a teminal on Linux or PuTTY on Windows and connect via SSH. [...]

I made sure I give the right id.
The make flash command failed but the below was able to complete the step:

From (https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender5-2019.cfg)

avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex

ld3300, 2019-12-07:
Does your Ender's Melzi board have a bootloader?  I am guessing Klipper uses it, but am not certain.  That was the first thing I had to do on mine to change firmware when I got it, though I started with the latest Marlin before finding Klipper.

srekan, 2019-12-07:
> Does your Ender's Melzi board have a bootloader? I am guessing Klipper uses it, but am not certain. That was the first thing I had to do on mine to change firmware when I got it, though I started with the latest Marlin before finding Klipper.

I am newbie to this hardware and firmware.
Here is what I did:
1. I installed Octopi on RaspberryPi then connected to the printer. (So far I am under the impression that Octopi is the bootloader - starts when powered on)
2. Accessing octoprint was fairly easy. I am able to control the prints from octoprint
3. Tried to install klipper
4. What happens after that is this page - I raised this issue

Can you please give some instructions to find out whether I have the bootloader on the Melziboard?

Arksine, 2019-12-07:
If its flashing then you have a bootloader.  The menuconfig looks correct.

Make sure that nothing else is trying to connect to the board, such as octoprint.  Check octoprint's serial port settings and make sure that /tmp/printer is selected.  You don't want it on AUTO trying to connect to the MCU.

[image]

If everything is good there, the only thing I can think of is that the bootloader isn't properly loading the program.  In that case you may have to flash a new one.

ld3300, 2019-12-07:
In case you want to try redoing the bootloader on the printer mainboard:

[Guide: How to Flash a Bootloader on Melzi Boards](https://letsprint3d.net/guide-how-to-flash-a-bootloader-on-melzi-boards/)

KevinOConnor, 2019-12-23:
I'm going to close this as it looks like the conversation has concluded.  Alas, it does not appear anyone knows the root cause of the failure to flash on this particular board.

-Kevin

closed 2019-12-23 by KevinOConnor