Klipper3d/klipper #2225: Internal error on command:"G1"

matyasf, 2019-11-28:
The error in the title comes up during a print. I am on the master branch, pulled around a day ago.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3903001/klippy.log)

As I see from my log this is the stack trace:
```
stepcompress o=2 i=-55704 c=1 a=0: Invalid sequence
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 219, in _process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 74, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/gcode.py", line 515, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/skew_correction.py", line 73, in move
    self.next_transform.move(corrected_pos, speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 160, in move
    self.toolhead.move([x, y, z + self.fade_target, e], speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 402, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 164, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 320, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 275, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/klipper/klippy/stepper.py", line 159, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
Dumping gcode input 50 blocks
```

matyasf, 2019-11-29:
The error is repeatable, it fails on the same spot for a certain .gcode file, I can send it if its useful.

KevinOConnor, 2019-11-30:
There appears to be an error in the extruder scheduling.  I am able to reproduce the problem from the info in the log.  However, it will likely take a few days to track it down.

-Kevin

KevinOConnor, 2019-12-03:
This should be fixed now (commit 988ee575).

Thanks.
-Kevin

matyasf, 2019-12-03:
Thanks for the fix!

closed 2019-12-13 by KevinOConnor

robustini, 2019-12-29:
Not fixed, i've the similar problem: #2334.

p3dstore, 2020-01-19:
`Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 233, in _process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 480, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 176, in move
    self.toolhead.move(split_move, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 409, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 180, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 169, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 327, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 280, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/klipper/klippy/stepper.py", line 167, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress`

Same issue here. On Gcode I have been printing all week. 7 times.

I'm pushing flows on my mosquito, and from what I've seen it might be max_extrude_cross section related?

[Fade Config 1-19-20.txt](https://github.com/KevinOConnor/klipper/files/4082882/Fade.Config.1-19-20.txt)
[klippy log 1-19-20.zip](https://github.com/KevinOConnor/klipper/files/4082883/klippy.log.1-19-20.zip)