Klipper3d/klipper #2226: Feature Request: ability to send M117 upstream to Octoprint (and other codes)

ld3300, 2019-11-29:
[klippy_log.zip](https://github.com/KevinOConnor/klipper/files/3903345/klippy_log.zip)

Currently I am using the Octoprint DisplayLayerProgress Plugin.  I am unable to send M117 commands, that are generated in my gcode_macros, upstream to Octoprint to have them displayed on the onscreen popup. 

I was hoping I could use the RESPOND extended g-code command to force a macro generated g-code to be issued upstream by Octoprint instead of at the Klipper level.  Please let me know if this is implemented and I have not found the proper combination of "RESPOND", "TYPE", and "MSG" formatting.  

If it is not currently possible, can a RESPOND TYPE (perhaps named "gcode") be added that will allow sending a command type response with a g-code payload?

Arksine, 2019-11-29:
Octoprint isn't capable of processing a response from Klipper directly as a gcode.  What you are looking for is an "action command".  I can see from your configuration that you are on to that, but the problem  you are facing is that support for the action command needs to be added on Octoprint's side, specifically from the plugin's developer.  So for example, lets say the developer of the DisplayLayerProgress plugin adds support for "action:updateprogress".  Then you could notify the plugin from klipper as follows:

```
[gcode_macro my_progress]
default_parameter_PROGRESS: 0
gcode:
  {printer.gcode.action_respond_info("action:updateprogress " + PROGRESS)}
```

ld3300, 2019-11-29:
Ahhh, yes, thank you.  I had it stuck in my head that that would be only a Klipper change, but yes it would require Octoprint to enable handling the message.  I was hoping to have a feature like this for far more than just M117 commands.  Maybe I can figure out the code enough to contribute to the Action Commands plugin to make it possible to receive custom commands with attached variables.

Thank you again.

ShohninDmitriy, 2019-12-01:
https://github.com/cesarvandevelde/OctoPrint-M73Progress

ld3300, 2019-12-07:
Needs to be implemented in Octoprint or via a plugin. Change not needed in Klipper

closed 2019-12-07 by ld3300