Klipper3d/klipper #2227: Ender3 SKR mini E3 v1.2 - lcd issues

jorgempy, 2019-11-29:
Hi! 

Im trying to migrate to the mini e3, and the only issue im running into is the lcd.
It sometimes work but most of the time it doesnt.
Somehow it works better with the lcd unscrewed from the ender3 frame.
My only idea is noise, but with a shorter ribbon the issue still happens.
I attach pictures of the issue.
[image]
[image]
[image]

also a short video https://youtu.be/lu8XgaTRjU0

i have read #2069 and tried with and without the aliases (both ways "work").

any ideas, things i could try? 

running on commit 73e84abaff7759cac0f40b52d5553406c1395e76 (latest master as of 29nov2019)

jorgempy, 2019-11-29:
Attached log [klippy.log](https://github.com/KevinOConnor/klipper/files/3905989/klippy.log)

BlackStump, 2019-11-29:
I have a SKR Mini E3 V1.2 and the Ender3 LCD works absolutely fine.
The major difference between your config and mine is I do not run a secondary mcu so just maybe that is where the problem rests.
Try disconnecting the secondary mcu and test.

KevinOConnor, 2019-12-13:
As noted, the description looks like noise.  It's certainly odd that you still have noise after replacing the cables, but we haven't received of any reports of general noise issues.

If you are a developer, it is possible to alter the timing settings in src/lcd_st7920.c and in klippy/extras/display/st7920.py to see if that has an effect.

-Kevin

closed 2019-12-30 by KevinOConnor