Klipper3d/klipper #2228: Cant flash SKR1.3

alexandrrusskih, 2019-11-30:
I try to install the klipper on skr1.3, but the board does not start. In menuconfig set
   * Enable extra low-level configuration options
        Micro-controller Architecture (LPC176x (Smoothieboard))
        Processor model (lpc1768 (100 Mhz)) --->
   * Target board uses Smoothieware bootloader
   * Use USB for communication (instead of serial)
  Nothing happens.
  Marlin 2.0 works fine on this board

ShohninDmitriy, 2019-11-30:
https://www.klipper3d.org/Installation.html

alexandrrusskih, 2019-11-30:
I did everything according to this instruction, but nothing helped

BlackStump, 2019-11-30:
did you copy the klipper.bin to the sdcard and rename to firmware.bin insert the sdcard into the skr and power up the board?
> after running "make", copy the generated "out/klipper.bin" file to a
> file named "firmware.bin" on an SD card and then restart the SKR
>  with that SD card.

unSinn, 2019-11-30:
Also make sure you have the right USB-Port configured  

Locate it on Octopi:
```
pi@octopi2:~ $ ls /dev/serial/by-id
usb-Klipper_lpc1768_1210010801A538AFCF66DE5BC12000F5-if00
```

And put in your configuration:
```
[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1768_1210010801A538AFCF66DE5BC12000F5-if00
```

glowtape, 2019-12-06:
I flashed an SKR 1.3 today that I just received a few days ago. Put firmware.bin onto SD card, power up the board, done. If you yank the SD card and mount it on your computer, and the file has been renamed to firmware.cur, it's been flashed. Then something else must be an issue.

alexandrrusskih, 2019-12-07:
Thanks to all. I just change sd card and now it works fine.

closed 2019-12-07 by alexandrrusskih