Klipper3d/klipper #2229: Klipper identifies itself, then fails when trying to print

ZathrasOne, 2019-12-01:
I appear to have everything setup correctly, and the OctoKlipper plugin is talking to the Klipper firmware, but when I actually try to print, it fails. I have also included the terminal output below...
[klippy.log](https://github.com/KevinOConnor/klipper/files/3907911/klippy.log)

Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6882b3d0, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Send: N0 M110 N0*125
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Recv: ok
Send: N1 M115*39
Send: N2 M21*18
Recv: ok FIRMWARE_VERSION:v0.8.0-154-g73e84aba FIRMWARE_NAME:Klipper
Recv: ok
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Send: N1 M107*36
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Changing monitoring state from "Printing" to "Cancelling"
Send: N2 M108*40
Send: N3 M84*28
Recv: ok
Recv: // Unable to connect
Send: N4 M104 T0 S0*37
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Send: N5 M108*47
Recv: ok
Send: N6 M140 S0*99
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Send: N7 M106 S0*96
Recv: ok
Send: N8 M108*34
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Changing monitoring state from "Cancelling" to "Operational"
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Send: M84
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Send: M108
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Send: M104 T0 S0
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Send: M140 S0
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Send: M106 S0
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Send: M84
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Send: M104 T0 S0
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Send: M140 S0
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Send: M106 S0
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Send: M84
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Send: M104 T0 S0
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Send: M140 S0
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Send: M106 S0
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect
Recv: ok
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: FIRMWARE_RESTART
Recv: ok
Recv: // Klipper state: Disconnect
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: RESTART
Recv: ok
Recv: // Klipper state: Disconnect
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: Status
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: // Klipper state: Not ready
Recv: !! Printer is not ready
Recv: ok
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
...

BlackStump, 2019-12-01:
https://github.com/KevinOConnor/klipper/issues/2223

ZathrasOne, 2019-12-02:
I rebuilt and reflashed the Klipper firmware and that appeared to fix the issue.

closed 2019-12-02 by ZathrasOne