Klipper3d/klipper #2230: End stop repeatability test (recommendation)

hermes3008, 2019-12-01:
I would like a way to find out standard deviation for endstops(could be and extension of probe repeatability)

KevinOConnor, 2019-12-01:
Looks like a dup of #2169.

-Kevin

Magoo0876, 2019-12-03:
Repeatability check

After calibrating the probe X, Y, and Z offsets it is a good idea to verify that the probe provides repeatable results. Start by homing the printer and then move the head to a position near the center of the bed. Navigate to the OctoPrint terminal tab and run the PROBE_ACCURACY command.

This command will run the probe ten times and produce output similar to the following:

Recv: // probe accuracy: at X:0.000 Y:0.000 Z:10.000
Recv: // and read 10 times with speed of 5 mm/s
Recv: // probe at -0.003,0.005 is z=2.506948
Recv: // probe at -0.003,0.005 is z=2.519448
Recv: // probe at -0.003,0.005 is z=2.519448
Recv: // probe at -0.003,0.005 is z=2.506948
Recv: // probe at -0.003,0.005 is z=2.519448
Recv: // probe at -0.003,0.005 is z=2.519448
Recv: // probe at -0.003,0.005 is z=2.506948
Recv: // probe at -0.003,0.005 is z=2.506948
Recv: // probe at -0.003,0.005 is z=2.519448
Recv: // probe at -0.003,0.005 is z=2.506948
Recv: // probe accuracy results: maximum 2.519448, minimum 2.506948, range 0.012500, average 2.513198, median 2.513198, standard dev

hth Robert

KevinOConnor, 2019-12-13:
I'm going to close this is favor of the existing #2169.

-Kevin

closed 2019-12-13 by KevinOConnor