Klipper3d/klipper #2231: BigTreetech TFT 24 not working

erik23de, 2019-12-01:
Hi,

Tried to set up Klippy on an skr 1.3 (from the example- configs), copy & pasted the Preprap GLCD
into the config.
Usually, testet with 3 Boards, Marlin and Repetier, the BigTreetech TFT 24 (and 35) works as a 12864 GLDC like a charm (First Display I'm really happy with).
But it does not with Klipper, Screen stays empty (but reset switch works). Wiring is okay, it works with Marlin.
What did I wrong?

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3908316/klippy.zip)

KevinOConnor, 2019-12-03:
I don't know of anyone that has attempted to use this device with Klipper.  It appears the device is trying to emulate an st7920 device.  As a guess, it may not fully emulate the device and thus not be compatible with Klipper.  (Klipper uses some features of the st7920 - such as the character generator - that other firmware don't use.)

-Kevin

erik23de, 2019-12-03:
Thanks for the info.

You should give it a chance - it's a good & cheap Display and we'll see more of that around soon.
With more conventional firmware, it works quiete easy with the widespread RepRapDiscount GLCD configuration.

In meantime, maybe I'm luckier with Klipper and the fysetc display - with wich I had no chance on Marlin or Repetier.

jalanjarosz, 2019-12-04:
@erik23de   I have a Fysetc MKS Mini display with the NeoPixel LED's working though a RAMPS shield on an Adafruit Grand Central M4.  Let me know if you need a hand with the config.

RomRider, 2019-12-06:
Same here with TFT35, if you switch back and forth between the touch mode and the marlin mode, it will ultimately display a bunch of garbage after playing with the button:
[image]

derBG, 2019-12-07:
Iam also interested. Does anyone have a config that shows anything? My TFT24 connected to skr 1.3 shows in the 12864 mode only a blue display.

matty-byte, 2019-12-16:
I have a TFT24, neither the emulated screen nor the touch screen will connect. The screen defaults to 250k baud rate which does not connect. Changing the baud rate to 115k does nothing and when rebooting it resets to 250k. BTT have guides for Marlin, but I have not been able to work out if any of these settings can be configured with Klipper.

KevinOConnor, 2019-12-17:
FYI, it appears that the tft24 (and similar) are emulating an st7920 chip:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/e180179aa945cfcf4a5af5c71581138d4d87225d/TFT/src/User/API/UI/ST7920_Simulator.c

That emulation does not include the character generator, so it is not compatible with Klipper.  However, if a developer is interested, it should be possible to add support to Klipper.

-Kevin

softkannan, 2020-01-23:
1.	BTT and MKS TFT both emulates host (like octoprint), no config. The incoming serial data is  GCODE commands and they listen for Ok and other serial command just like octoprint. Marlin tackled these displays by having more than one serial port can act as host.

2.	Anycubic TFT display also sends serial texts on high level but key difference is Anycubic TFT is not acting like host, its serial commands needs to be translated (requires intermediator).

3.	All the above 3 displays offer no config, means we cannot add new buttons or any other menus without compiling display firmware.

4.	to support above type of displays, on klipper we need
      **on mcu side**
       1. Generic serial text send
       2. Generic serial  text recev
      **on klippy side**
       1. Generic gcode injector to queue and response, it has to emulate klippy virtual serial port functions, like sending ok, busy etc. 
       2. Once above “Generic gcode injector to queue” is ready then we can customize for Anycubic TFT type displays.

**Benefit**
        - It makes klipper deployment easy, no hardware modification required, except plugging in rpi. 
        - Above approach is not specific to any display (except anycubic, I believe that must be done on private), which also preserve future expandability.
**Cons**
	It might increase some traffic, but I guess impact on performance is minimal.

Is above approach make sense ?

alagerqvist, 2020-02-23:
What klipper display settings are you using while trying to get the TFT24/35 running.

This is my interpretation of how to set it up.
[display]
lcd_type: st7920
cs_pin: PD10
sclk_pin: PG2
sid_pin: PD11
encoder_pins: PG10,PF11
click_pin: PA8
# Klipper     Marlin
# cs_pin == LCD_PINS_RS
# sclk_pin == LCD_PINS_D4
# sid_pin == LCD_PINS_ENABLE
# encoder_pins == BTN_EN1, BTN_EN2
# click_pin == BTN_ENC
[output_pin beeper]
d4_pin: PG2
d5_pin: PG3
d6_pin: PG6
d7_pin: PG7

[output_pin beeper]
pin:  PG4

[gcode_macro M300]
default_parameter_S=1000
default_parameter_P=100
gcode:  SET_PIN PIN=beeper VALUE=1
        G4 P{P}
        SET_PIN PIN=beeper VALUE=0

Tested the M300 to get a beep from the display =D

DeviousPenguin, 2020-02-26:
@alagerqvist I agree serial communication with the screen in 'touch' mode would be nice, I'm currently experimenting with using a USB2TTL adaptor connected from the RPi directly to the Display, and while it connects, I'm unable to get it to do anything useful so far.

This way the USB link betweek Klipper and the MCU is totally unhindered, the Screen is on a different bus, and can even be a different baud rate, but the wiring for this is less clean if your RPi is not mounted to your printer.

Can I ask if you managed to get the 'marlin mode'/'12864 simulator' mode to get anything to appear on the screen? Or is it only the Beep that works?

alagerqvist, 2020-02-26:
@DeviousPenguin  
I have tried a bit of debugging but my problem is that I don't receive ANY serial communication from klipper, it is probably due to my display set-up in klipper, but the problem is that the documentation for setting up display for klipper is close to none.
In my previous comment I tried to describe my guess on how to map Marlin configurations to klipper as marlin configurations are easy to come by for any hardware.

So if someone could confirm/oppose the below I'd be happy: (Do I lack something?)
Klipper <==> Marlin:
cs_pin <==> LCD_PINS_RS
sclk_pin <==> LCD_PINS_D4
sid_pin <==> LCD_PINS_ENABLE
encoder_pins <==> BTN_EN1, BTN_EN2
click_pin <==> BTN_ENC
 
I have started on an implementation of "full 7920 display emulation" for the TFT24/35 (or other), but I really can't test until I get serial working against klipper.
It look quite straightforward to change the 7920 code in klipper but I'm not sure that is the way to go... (but serial multiplexing is a must have....)

DeviousPenguin, 2020-02-29:
@alagerqvist I may be wrong but 7920 display emulation mode doesn't need serial in order to work.

I had a SKR v1.4 board turn up today, once I get it plugged up and flashed with Klipper, I'll see if I can confirm your pinouts and get some form of functionality with the TFT24

alagerqvist, 2020-02-29:
Hi,

I was a bit unclear, when I said serial was'nt working I actually meant the
SPI communication to the display.
So what I mean is that the screen gets no 7920 instructions over the
configured pins, I added on screen debug info to see.

alagerqvist, 2020-02-29:
@DeviousPenguin 
I was a bit unclear, when I said serial wasn't working I actually meant the SPI communication to the display.  So what I mean is that the screen gets no 7920 instructions over the configured pins, I added on screen debug info to see if the screen get any data. 

The below might have some impact: (This is the setting that works against the display using Marlin)
    #define BOARD_ST7920_DELAY_1 DELAY_NS(96)
    #define BOARD_ST7920_DELAY_2 DELAY_NS(48)
    #define BOARD_ST7920_DELAY_3 DELAY_NS(600)
I have no idea if possible/how to set those delays for Klipper...

alagerqvist, 2020-02-29:
@KevinOConnor 
I took a look at the low-level specification for communicating with the 7920 and what Klipper does seems a bit of the spec for TS7920.
Setting the clock-pin to low before setting the data and then pulling clock up seems correct according to the data sheet. (So data is read on rising clk).
(According to the spec. page 38 "Serial interface timing diagram   MPU write data to ST7920")

The timing of Klipper looks strange and the clk is toggled instead of set.

Marlin does this: (Works with BTT code which isn't real 7920 HW) 
  WRITE(ST7920_CLK_PIN, LOW);             ST7920_DELAY_1; \
  WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
  WRITE(ST7920_CLK_PIN, HIGH);            ST7920_DELAY_3; \

Klipper does:
        if (data & 0x80) {
            gpio_out_toggle(sid);
            data = ~data;
        }
        ndelay(200);
        gpio_out_toggle(sclk);
        ndelay(200);
        data <<= 1;
        gpio_out_toggle(sclk);

KevinOConnor, 2020-03-02:
I'm not aware of any issues with the Klipper bit banging interface to the st7920.  What you're looking at is the low-level toggling that is done - it's different from Marlin, but I don't believe it is incorrect.

Separately, when I last looked at the BTT firmware running on the display (https://github.com/KevinOConnor/klipper/issues/2231#issuecomment-566371883) it did not support the st7920 commands that Klipper issues.  I'm pretty sure that's the root of the issue.

-Kevin

alagerqvist, 2020-03-02:
Hi,

I was actually checking against the hardware sheet, I understand that
Marlin isn't a golden standard of implementation for this.

Anyway, I have attached a picture from the specification showing what I
mean, there is also a timing table that is dependant on the Vcc 3.3V or 5V
for the TS7920.

My point is that trying to follow the hardware spec. seems to make the
display software fail for Klipper.

Cheers/
Anders Lagerqvist

4noxx, 2020-04-06:
hi, are there a solution to run the bigtreetech tft-display (perhaps in marlin mode)
in klipper?
i use a SKR E3 MINI, and no Display runs on my Board (test bigtreetech tft35 and an reprap graph display and my old anet a8 display).

JohnEdwa, 2020-04-06:
AFAIK, no. Klipper requires the display to support the character generator, which the BTT emulator doesn't have. There is [an open issue on the firmware page](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/290) about it, but nobody has implemented it yet.   
Also, as it's not needed by Marlin, only by Klipper and other misc projects, the chances are unless someone from here goes to implement it, it never will.

DeviousPenguin, 2020-04-16:
> Also, as it's not needed by Marlin, only by Klipper and other misc projects, the chances are unless someone from here goes to implement it, it never will.

I'm not the most experienced with C++ and Arduino type code, but can do the basics. Would you know the best place to start with implementing the character generator?

trevjonez, 2020-04-16:
> I'm not the most experienced with C++ and Arduino type code, but can do the basics. Would you know the best place to start with implementing the character generator?

From the other thread just recently:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/290#issuecomment-610286699

It sounds like they may not emulate the char generator, but use font files that are already present in the firmware.

DeviousPenguin, 2020-04-17:
@trevjonez thanks it does look like BTT may be adding the char gen, by the looks of that thread, so we may not need to re-invent the wheel

dushyantahuja, 2020-05-09:
I've just got this TFT (a few minutes back), and see that there's a USB port behind it. Would it be possible to connect this to the pi and simply link the two serial port together (one from the TFT and /tmp/printer) - so that any commands the touch screen sends would get sent to klipper.

I'm no Linux expert - but it might be possible using socat. Will have to try once I get my board installed and have the printer running again. 

http://technostuff.blogspot.com/2008/10/some-useful-socat-commands.html

JohnEdwa, 2020-05-09:
That port is only for connecting a USB flash drive to the printer instead of using the SD card slot. The serial port is on a different connection entirely, as that is indeed how the display communicates using the "touch screen mode" with the mainboard: purely by sending and receiving Gcode commands, just like OctoPrint does.

Connect that display to one of the serial ports on your Raspi instead of the mainboard and instruct Klipper to listen to that port, and you could, in theory, replace Octoprint with that touchscreen. With the caveat that it's almost guaranteed to use some Marlin specific commands that won't work with Klipper.

dushyantahuja, 2020-05-10:
Aah - too bad. Turned out when I shorted my previous control board, it also took the TMC drivers with it :(. Now to wait for the new ones to arrive, before I can experiment further. More thoughts:

1. As Kevin mentioned, port Klipper directly to TFT and use it as another microcontroller. Since I have no experience in this, I'll leave it to the experts
2. Connect to ESP8266 - and then use the Octoprint API to get / set data. It shouldn't be too difficult, let me see what I can do.

HenrikPPersson, 2020-05-22:
Hi! New owner of Bigtreetech TFT24 and first poster on Github.
I looked through the source of the TFT24 firmware and discovered that the display only emulate the ST7920 graphics mode. Not the text or glyph mode. After modifying the st7920.py in klipper I sort of got it working.

When I start up the screen, it is blank, part from the "LCD12864 Simulator" text at the top. Once I switch over to Serial Touch Screen mode and then back again it starts to show things. With the patch below, it seems to be working (part from the animations).

```diff
@@ -126,7 +126,11 @@ class ST7920:
         if x + len(data) > 16:
             data = data[:16 - min(x, 16)]
         pos = [0, 32, 16, 48][y] + x
-        self.text_framebuffer[pos:pos+len(data)] = data
+        # self.text_framebuffer[pos:pos+len(data)] = data
+        for c in data:
+            for i, bits in enumerate(bytearray(font8x14.VGA_FONT[ord(c)])):
+                self.write_graphics(x, y, i, [bits])
+            x=x+1
     def write_graphics(self, x, y, row, data):
         if x + len(data) > 16:
             data = data[:16 - min(x, 16)]
```

trevjonez, 2020-05-22:
what kind of increased load does that put on the MCU? seems like a decent short term workaround while we wait for btt to finish the char generator.

HenrikPPersson, 2020-05-22:
So far I haven't run it more than a couple of hours, so long time effects are unknown. The solution for uc1701 is very similar to this, so the the MCU load shouldn't be much worse than with that one.

Right now, I have it hooked up to a spare controller board (Trigorilla) and running klipper directly on my development WS (Fedora). I will connect it to one of my printers and see how it behaves in real life.

nklrk, 2020-05-24:
FYI: Yesterday (23.05.2020) BTT added support for klipper (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/e6d8218dc20d4d5d4b3a2700c0e3fc26bb4059ed)

alagerqvist, 2020-05-24:
Hi,

Sorry, I haven't tried for a long while.
I think the low level bit banging timing is the root problem then the
character emulator needs to be implemented.

/Anders

dushyantahuja, 2020-05-25:
Hmmm - just uploaded the latest firmware to my TFT24 (compiled from the latest source) - and have the following in my config:

```
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"

[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
#kill_pin: ^!EXP2_8
```
Still doesn't work - I just get a blank screen with "LCD12864 simulator" on top
Has anyone got it working - can you please advise if I've got the pins wrong, or maybe something else.

HenrikPPersson, 2020-05-25:
There seems to be something wrong with the initialization of the ST7920 mode in the TFT24-display. I have to switch over to Touch mode and then back to Marlin mode after i power on. Otherwise the display stays black.

dushyantahuja, 2020-05-25:
Got it working - needed to change
```
SPI_ReEnable(!!(GPIOB->IDR & (1<<13))); to SPI_ReEnable(1)
```
in https://github.com/Msq001/BIGTREETECH-TouchScreenFirmware/blob/236c497d6050215b9674f6f78ffe0d581a834417/TFT/src/User/Hal/stm32f10x/spi_slave.c#L119

There's also a binary in the pull request that you can use directly:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/705

xkuyax, 2020-05-30:
Hi @dushyantahuja thanks for you work on this! 
Could you help me to get my LCD working?

I have the following setup:
Klippy v0.8.0-526-gf0c394de
Skr Pro 1.1 
TFT35 V3 with the newest Firmware (cloned fresh from git) and i did the change in spi_slave 

At first i attached the TFT Dupont Cable, but i disconnected that (as it interfered with klipper)

Now i only have exp1 and exp2 connected and the following in my config:
[board_pins]
aliases = 
	EXP1_1=PG4, EXP1_3=PD11, EXP1_5=PG2, EXP1_7=PG6, EXP1_9=<GND>,
	EXP1_2=PA8, EXP1_4=PD10, EXP1_6=PG3, EXP1_8=PG7, EXP1_10=<5V>,
	EXP2_1=PB14, EXP2_3=PG10, EXP2_5=PF11, EXP2_7=PF12,  EXP2_9=<GND>,
	EXP2_2=PB13, EXP2_4=PB12, EXP2_6=PB15, EXP2_8=<RST>, EXP2_10=PF13
(taken from the example of the skr pro)

[display]
lcd_type = st7920
cs_pin = EXP1_4
sclk_pin = EXP1_5
sid_pin = EXP1_3
encoder_pins = ^EXP2_3, ^EXP2_5
click_pin = ^!EXP1_2

But i only see the LCD12864 when selecting the marlin mode...
Anything special i need to configure in the config.ini or something else?

dushyantahuja, 2020-05-30:
> EXP2_2=PB13, EXP2_4=PB12, EXP2_6=PB15, EXP2_8=, EXP2_10=PF13

The firmware is not perfect yet. Do you have a knob on your TFT? Sometimes
I see the blank screen as well, but if I press the knob a couple of times,
the menu comes up.

Try changing to touch screen mode and then back to classic mode. That
should work.

Last option would be to compare the pins on the SKR 1.1 and 1.4 - if they
have changed, you will have to change the pin definitions as well.

HenrikPPersson, 2020-05-30:
Hi.
I also run with at SKR Pro 1.1 and the pin definition is the same as mine. As @dushyantahuja wrote, try switching over to Touch mode and then back again. That is what I'll have to do to make it works after a reset/reboot.

SrGeek, 2020-06-19:
Any progress on this front?  I have a SKR mini E3 and only have TFT available to me.

cata747, 2020-06-19:
Hi. Mine worked only using TFT35 folder NOT from unified menu material theme.  Did the mods from [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/705#issue-422238077](url) 
and switch back and forth from touch mode to marlin mode. 
I have skr 1.3 and TFT35 and in printer.cfg I have
[display]
lcd_type: st7920
cs_pin: P1.19
sclk_pin: P1.20
sid_pin: P1.18
encoder_pins: ^P3.26, ^P3.25
click_pin: ^!P0.28
I hope it helps someone.

SrGeek, 2020-06-19:
@cata747 I only have one EXP1 and a TFT connection on my board.  There is no EXP2, thats where the encoder pins are at.

SrGeek, 2020-06-19:
can I use a second mcu for my screen  i have an extra re-arm board it has exp1 & exp2.

cata747, 2020-06-20:
@SrGeek There are config files for skr mini e3 for compiling the firmware. You can start there.
And some google [https://www.reddit.com/r/BIGTREETECH/comments/dtl8id/skr_mini_e3_12_klipper_configguide_including/](url)

KevinOConnor, 2020-08-19:
Any further updates on this issue?  What's the current state of the BTT TFT on Klipper?

-Kevin

trevjonez, 2020-08-20:
I've been running a custom build of the BTT firmware on my klipper machines to _some_ success.
As best I can tell the issue is purely on the BTT simulator logic because a real st7920 works fine. And their issue is still open and pinned for the char generator.

One nice thing is by doing a custom logo.bmp file I get a klipper logo for the splash screen on firmware restart. But the overall experience is far less than what has become consistent for klipper.

dsbaars, 2020-09-05:
> Any further updates on this issue? What's the current state of the BTT TFT on Klipper?

Regarding **"Marlin Mode" / LCD12864 emulation mode**:
My TFT35 E3 V3.0 came with a firmware from June, this did not seem to work with Klipper. 
I compiled my own firmware, which works okay. Sometimes there are artefacts and sometimes you need to go to "touch mode" first and then to Marlin mode to make it work.
You can download my firmware and use it **at your own risk** from:
https://mega.nz/file/jbBUQKIa#xXKu3-K9MmVkF6lSxHjqni2rWLsxexGOXU1E_LxA-hc

I compiled this for an TFT35 **E3** V3.0 used with an Ender 3 (2020) with BTT SKR MINI E3 v2.0 firmware (note that there are two versions of the TFT35 V3.0, this one is for the fit-in replacement of the Ender 3 LCD, hence the "E3" addition).
 
The relevant part from `printer.cfg` to use this
````YAML
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>

[output_pin beeper]
pin: EXP1_1

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
````

While reading replies in several issues in the [BTT TFT Firmware repository](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/281), I assume adding "read-only" support for "**Touchscreen mode**" should be quite simple at least the following two ways:
1) **IF** OctoPrint is able to "mirror" the data sent to Klipper to another serial port, connect your TFT directly to the GPIO pins of your Raspberry Pi (RX2/TX2/GND)
2) Re-send the commands which are sent from Octoprint to Klipper over the TFT interface of the SKR (E3) mainboards, for the [BTT SKR MINI E3 v2.0](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf) this would be pins RX2/TX2

alagerqvist, 2020-09-05:
Hi,

Sorry, I haven't tried anymore as I'am using marlin only.
I started on a emulator mode but then covid happened and shifted focus on
helping with visor prints etc.
/Anders

lilasiian8, 2020-09-10:
> Regarding **"Marlin Mode" / LCD12864 emulation mode**:
> My TFT35 E3 V3.0 came with a firmware from June, this did not seem to work with Klipper.
> I compiled my own firmware, which works okay. [...]

I can confirm that this successfully worked for me. Thank you for sharing this!

artheoday, 2020-09-23:
How do I do this? Is there a tutorial to do this?

> I can confirm that this successfully worked for me. Thank you for sharing this!

DroneMang, 2020-10-07:
The TFT24 display doesn't appear to work with Klipper.  I just tried this with the precompiled firmware from BTT and by compiling it, with the unified menu and standard, and nothing shows up other than LCD12864 simulator.  I have an SKR pro.

derBG, 2020-11-01:
We are waiting for the source code and binary update by BTT

basha318i, 2020-11-16:
Hello, 
I managed to get it working after changing SPI_ReEnable(!!(GPIOB->IDR & (1<<13))); to SPI_ReEnable(1)
but again still with some artefacts. this is how it looks on a BTT TFT35 V3.0 connected to an SKR 1.4 Turbo.
[image]
I'm not sure if it's just me or everyone is having this, but it looks really dull; like the font needs to be changed & made smaller; also I believe there's a better way of arranging stuff, similar to the Marlin Simulation, maybe?

derBG, 2020-12-08:
Meep

DroneMang, 2020-12-08:
I gave up and went with the stock reprap full display for now, if this
works that would be great! I got it working with the TFT 24 but I would
have to restart the firmware somewhere between 5 to 10 times to get the
display to actually come up.

gkatev, 2021-01-03:
@github-actions Let's keep it open? :)

MarsArtis, 2021-01-12:
> Regarding **"Marlin Mode" / LCD12864 emulation mode**:
> My TFT35 E3 V3.0 came with a firmware from June, this did not seem to work with Klipper.
> I compiled my own firmware, which works okay. [...]

Can you tell If this custom firmware may be used on TFT24 too, please?

teeminus, 2021-01-13:
I just posted a reply to https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/290: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/290#issuecomment-759543165

I think I fixed the issue, at least for me the TFT35v3 is now working with klipper. I also created a very stripped down and minimalistic version of the firmware that has just the ST7920 emulator left as I only use that mode.

eremid, 2021-01-14:
Thanks @teeminus ! So if I want to test on my TFT35v3, i need the firmware in your fork repository and the config bellow ?

```
[display]
lcd_type: st7920
cs_pin: P1.19
sclk_pin: P1.20
sid_pin: P1.18
encoder_pins: ^P3.26, ^P3.25
click_pin: ^!P0.28
```

Thanks

teeminus, 2021-01-14:
@eremid You can either use the firmware from the fork (you have to compile it yourself) or you use the precompiled binary from the stripped down firmware: https://github.com/teeminus/NoTouchScreenFirmware

I don't know if the klipper config is correct as I have a different mainboard. At least all required config entries are present, so if the pins are correct the display should work.

MarsArtis, 2021-01-14:
> I just posted a reply to [bigtreetech/BIGTREETECH-TouchScreenFirmware#290](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/290): [bigtreetech/BIGTREETECH-TouchScreenFirmware#290 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/290#issuecomment-759543165)
> 
> I think I fixed the issue, at least for me the TFT35v3 is now working with klipper. I also created a very stripped down and minimalistic version of the firmware that has just the ST7920 emulator left as I only use that mode.

Have tried your custom Firmware with my **BIGTREE_TFT24_V1_1**.
After the flashing is done I get this error:

**Illegal FLASH APP!**

Sellecta, 2021-01-14:
I tried to use it on tft24, the screen is full of artifacts, maybe I got something wrong with the pins [https://photos.app.goo.gl/1nHccEQo3xTKoiyY6](url)

dsbaars, 2021-01-14:
@MarsArtis @Sellecta the firmware I provided is for TFT35 (E3) only. Like @teeminus said, you have to compile it yourself.

[55 more replies]

closed 2021-03-10 by github-actions[bot]