Klipper3d/klipper #2232: Location of Screws Test Adjust Must Be After Probe?

15goudreau, 2019-12-01:
So I kept running into an error in the screws_test_adjust section of my config 
`Recv: // Internal error during connect: ScrewsTiltAdjust instance has no attribute 'gcode'
Recv: //
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: // Klipper state: Not ready
Recv: !! Internal error during connect: ScrewsTiltAdjust instance has no attribute 'gcode'`

Once I moved that section below my probing sections it started working. My config is attached. Is this intentional or a bug?

> # This file contains common pin mappings for the BIGTREETECH SKR V1.3
# board. To use this config, the firmware should be compiled for the
# LPC1768.

# See the example.cfg file for a description of available parameters.

#######################################################################
#                                                                     #
#                        Main Printer Settings                        #
#                                                                     #
#######################################################################

[printer]
kinematics: cartesian
max_velocity: 350
max_accel: 2500
max_z_velocity: 40
max_z_accel: 2000
square_corner_velocity: 10

[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1768_0E80FF1006083DAFE3A1715CC02000F5-if00

[idle_timeout]
# 1 day
timeout: 86400

[tmc2208 stepper_x]
uart_pin: P1.17
microsteps: 16
run_current: .9
hold_current: 0.7
stealthchop_threshold: 0
#interpolate: true

[stepper_x]
step_pin: P2.2
dir_pin: P2.6
enable_pin: !P2.1
step_distance: 0.0125
#   80 steps per mm - 1.8 deg - 1/16 microstepping
endstop_pin: P1.29
position_min: 0
position_endstop: 0
position_max: 320
homing_speed: 100
homing_retract_dist: 5

[tmc2208 stepper_y]
uart_pin: P1.15
microsteps: 16
run_current: .9
hold_current: 0.7
stealthchop_threshold: 0
#interpolate: true

[stepper_y]
step_pin: P0.19
dir_pin: !P0.20
enable_pin: !P2.8
step_distance: 0.0125
#   80 steps per mm - 1.8 deg - 1/16 microstepping
endstop_pin: P1.27
position_min: 0
position_endstop: 0
position_max: 315
homing_speed: 100
homing_retract_dist: 5

[tmc2208 stepper_z]
uart_pin: P1.8
microsteps: 16
run_current: .5
hold_current: .4
stealthchop_threshold: 0

[stepper_z]
step_pin: P2.13
dir_pin: P0.11
enable_pin: !P2.12
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_endstop: 0.5
position_max: 400
position_min: -50

[tmc2208 stepper_z1]
uart_pin: P1.10
microsteps: 16
run_current: .5
hold_current: .4
stealthchop_threshold: 0

[stepper_z1]
step_pin: P0.22
dir_pin: P2.11
enable_pin: !P0.21
step_distance: .0025

#######################################################################
#                                                                     #
#                        Extruder Settings                            #
#                                                                     #
#######################################################################

[tmc2208 extruder]
uart_pin: P1.1
microsteps: 16
run_current: 1.0
hold_current: 0.8
stealthchop_threshold: 0

[extruder]
step_pin: P0.1
dir_pin: !P0.0
enable_pin: !P0.10
step_distance: .010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.24
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
min_temp: 0
max_temp: 300

[verify_heater extruder]
check_gain_time: 1000

#######################################################################
#                                                                     #
#                        Bed Settings                                 #
#                                                                     #
#######################################################################

[heater_bed]
heater_pin: P2.5
sensor_type: ATC Semitec 104GT-2
sensor_pin: P0.23
control: pid
pid_Kp: 47.690
pid_Ki: 1.556
pid_Kd: 365.338
min_temp: 0
max_temp: 130

#	BED_MESH_CALIBRATE
[bed_mesh]
speed: 200
horizontal_move_z: 10
min_point: 35,5
max_point: 315,305
probe_count: 5,5
#fade_start: 0.5
fade_end: 10.0
split_delta_z: .01
move_check_distance: 3
#mesh_pps: 0,0
algorithm: bicubic
relative_reference_index: 12
#bicubic_tension: .02

[bed_screws]
screw1: 25,55
screw2: 280,55
screw3: 280,295
screw4: 25,295

[z_tilt]
z_positions:
	-30,150
	330,150
points:
	40,150
	305,150
speed: 200
horizontal_move_z: 5
retries: 5
retry_tolerance: 0.02
horizontal_move_z: 10

#######################################################################
#                                                                     #
#                        Fan Settings                                 #
#                                                                     #
#######################################################################

[fan]
# Print cooling fan
pin: P2.4
#   D9 on mcu_xye
#kick_start_time: 0.500
#cycle_time: 0.001
#fan_speed: 1.0

[heater_fan hotend_fan]
# Hotend fan
fan_speed: 1.0
pin: P2.3
#   D9 on mcu_z
kick_start_time: 0.500
heater: extruder
heater_temp: 50.0

#######################################################################
#                                                                     #
#                        Probe Settings                               #
#                                                                     #
#######################################################################

[bltouch]
sensor_pin: P1.25
control_pin: P2.0
x_offset: -25.0
y_offset: 20.0
z_offset: 1.9
speed: 20
samples: 2
sample_retract_dist: 5
#samples_result:
#samples_tolerance:
#samples_tolerance_retries:
pin_up_touch_mode_reports_triggered: False

[safe_z_home]
home_xy_position: 155,160 # Change coordinates to the center of your print bed
speed: 50
z_hop: 10                 # Move up 10mm
z_hop_speed: 5

[screws_tilt_adjust]
screw1: 50,35
screw1_name: front left screw
screw2: 305,35
screw2_name: front right screw
screw3: 305,275
screw3_name: rear right screw
screw4: 50,275
screw4_name: rear left screw
horizontal_move_z: 10
speed: 150
screw_thread: CW-M3

#######################################################################
#                                                                     #
#                        Macro Settings                               #
#                                                                     #
#######################################################################

[gcode_macro PRINT_START]
#   Use PRINT_START for the slicer starting script - PLEASE CUSTOMISE THE SCRIPT FOR YOUR SLICER OF CHOICE
gcode:
	bed_mesh_clear;
    M117 Homing...                 ; display message
    G28                            ; home all axes
    G1 Z20 F3000                   ; move nozzle away from bed
    M117 Preheat (Print)           ; display message
	z_tilt_adjust;
	G28;
#	bed_mesh_calibrate;
	G92 E0;
	G0 X40 Y20 Z0.3 F2000;
	G0 E15 F500;
	G0 X200 Y22 F2500 E30;
	G0 X200 Y30 F5000;
	G0 X40 Y30 F2500 E35;
	G92 E0;
	G1 Z4.0 F3000;
	
[gcode_macro PRINT_END]
#   Use PRINT_END for the slicer ending script - PLEASE CUSTOMISE THE SCRIPT FOR YOUR SLICER OF CHOICE
gcode:
    M400                           ; wait for buffer to clear
    G92 E0                         ; zero the extruder
    G1 E-2.0 F3600                 ; retract
    G91                            ; relative positioning
    G0 Z1.00 X20.0 Y20.0 F20000    ; move nozzle to remove stringing
    M104 S0                        ; turn off hotend
    M140 S0                        ; turn off bed
    M106 S0                        ; turn off fan
    G1 Z2 F3000                    ; move nozzle up 2mm
    G90                            ; absolute positioning
    G0  X150 Y300 F3600            ; park nozzle at the center rear
    M117 Finished!                 ; display message	

########################################
# EXP1 / EXP2 (display) pins
########################################

#[board_pins]
#aliases:
    # EXP1 header
#    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
#    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
#    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
#    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"

# See the sample-lcd.cfg file for definitions of common LCD displays.

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 23.383
#*# pid_ki = 0.987
#*# pid_kd = 138.546

So if I moved it into my "bed" section it throws that error.

15goudreau, 2019-12-01:
log file here 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3908634/klippy.log)

KevinOConnor, 2019-12-03:
Should be fixed now (commit a8ad1ae7).

Thanks,
-Kevin

closed 2019-12-03 by 15goudreau