Klipper3d/klipper #2233: horizontal lines on print

gkyle840, 2019-12-02:
I am getting horizontal lines on my prints pretty badly. I have adjusted and replaced almost everything on the printer and have narrowed it down to either something in octoprint or klipper.   the issue didnt start until I switched from Marlin to klipper and installed a genuine bltouch.

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3909357/klippy.log.zip)

Magoo0876, 2019-12-03:
Hi,
Its very hard without a bit more of information to help. Can you please describe your setup and add some photos of your lines. But the first advice is to update the software and firmware and adapt your config to run the new version. The config looks like a BMG direct extruder, the thing I've found out quickly was a very slow z axis. max_z_velocity = 5. I run my ender 3 dual z on 40 and the double of the velocity.

# how to update klipper:
# connect the host via ssh  
sudo service klipper stop
cd ~/klipper
git pull
sudo service klipper start

# how to update firmware
sudo service klipper stop
cd klipper
make menuconfig 
make clean
make
make flash FLASH_DEVICE=/dev/serial/by-id/usb-your_id_bla_bla_.....
sudo service klipper start

After the first connect you will get some errors because of the old configuration. 
As an an example pressure_advance_lookahead_time = 0.01 has to be removed

-Robert

gkyle840, 2019-12-04:
I am running an ender2 with a bmg D/D extruder and V6 all metal hotend. it has a linear rail on the Y axis.  here are my printer settings kinematics: cartesian
max_velocity: 300
max_accel: 1000
#1500
max_z_velocity: 5
#5
max_z_accel: 100
#100
square_corner_velocity: 4

gkyle840, 2019-12-07:
[image]

KevinOConnor, 2019-12-09:
It doesn't look like you are running code from this repo.  You'll need to pull the latest pristine code and retry in order for us to assist.  (See the FAQ for upgrade details.)

Otherwise, you can do a web search for "3d printer z banding" to find common causes of Z banding and their solutions.

-Kevin

johngianni, 2019-12-10:
@gkyle840 are you referring to the white cube? I've got a similar behavior that I will document as soon as I'm home tonight. I was guessing that it was a matter of printing very fast (175mm/s) and cheap linear rails on the x axis, but your thread caught my attention. Using a E3D V6 as well, with Flex3Drive G5 extr.

johngianni, 2019-12-11:
Here are my prints:
[image]

gkyle840, 2019-12-12:
@KevinOConnor  
here is my most recent log after updating.
[11klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3953838/11klippy.log.zip)

gkyle840, 2019-12-12:
@johngianni 
yes I am referring to the white cube.  I have printed them between 18 and 65mm/s and the results are about the same for me.  I am nowhere near trying it at 100+mm/s like you though.

johngianni, 2019-12-12:
Hi, FYI I've almost solved my imperfections by slightly correcting the tension of the screws that hold the G5 extruder together and grease the gears, + reducing by a 25% the pressure advance, + reducing by 5% the flow.

KevinOConnor, 2019-12-30:
I'm going to close this issue as it looks like the conversation has concluded.  As near as I can tell, this appears to be a typical "z banding" issue and not a software issue.  (Separately, the latest log still does not show code versions from this repo.)

-Kevin

closed 2019-12-30 by KevinOConnor