Klipper3d/klipper #2235: [Feature Request] PINDA thermistor z compensation

imhhawk, 2019-12-02:
I am using a PINDA V2 on a RepRap I built and love its leveling capabilities with Klipper. Prusa firmware uses the PINDA's built in thermistor to compensate for sensor drift caused by different temperatures. I can add a custom thermistor on an extra thermistor pin (RAMPS 1.4), but I can't (as far as I know) do z compensation based on it. Is there any way temperature compensation like Prusa's could be added? Or is there a way I can do it without a new feature?

Thanks!
(On mobile rn, don't have a log to attach, sorry)

KevinOConnor, 2019-12-02:
Looks like a dup of #1611.  FWIW, I think everyone that has worked on the pinda has concluded that the temperature compensation isn't particularly useful.

-Kevin

Magoo0876, 2019-12-03:
Maybe that helps: 

https://github.com/PrusaOwners/prusaowners/wiki/Klipper

There is a fork for Prusa's and they added the compensation like that.

[probe_temp]
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PF3
t_offsets:
 35.0, 0.0
 40.0, 0.02
 45.0, 0.06
 50.0, 0.120
 55.0, 0.2
 60.0, 0.3

I've never tried that, so would be nice to know if it does the job.
My solution was to move the pinda2 away from the bed or simply probe on the cold bed. 

-Robert

Arksine, 2019-12-04:
The probe_temp module works but there are some things about the module that are not ideal so I did not PR it.  FWIW, I have an alternative module that I will likely create a PR for in the future.  The module was not designed specifically for a Pinda however it would work on a Pinda.  It will need more testing before its ready.

imhhawk, 2019-12-07:
@KevinOConnor Apparently I'm bad at searching! Completely missed that issue.

@Magoo0876 Thanks, I'll take a look.

@Arksine I'll try probe_temp for now, I'd be interested to try your alternative.

closed 2019-12-07 by imhhawk

rafaljot, 2020-03-25:
Some months ago I made this:
https://github.com/KevinOConnor/klipper/pull/1611
but finally I found simple solution. Just move up Z 50 cmm above the bed during heating. This helps to keep sensor cool.

ripxorip, 2020-07-13:
> Some months ago I made this:
> #1611
> but finally I found simple solution. Just move Z up 50 cm above the bed during heating. This helps to keep sensor cool.

Are you sure you mean 50cm? My Z is only 21cm so I would have to detach the hotend