Klipper3d/klipper #2236: New MKS Gen L UART support for TMC2208

holmlrs, 2019-12-02:
Dear all,

I just switched to the new MKS Gen L V2.0 in my kossel printer. I'm using TMC2208 drivers in standalone mode, but I want to use them in uart mode. (they work fine in standalone mode) So, I removed the MS1, MS2 and Uart pin, bridged the UART pad to the NC pad (MKS L V2.0 has UART jumper connected to the NC pin), and bridged the two solderpads on the bottom to connect the uart pin. To the best of my knowledge, this mod connects the uart line of the tmc chip to the NC pin and removes pins that are unused in UART mode.

Then I added the necessary sections to the printer.cfg (attached below), and put in the UART jumper that connects the NC pin (now UART) to the ATMEGA2560. I checked the connection; the UART pins are connected to the pins stated in the datasheet (analog9-analog12) (provided below)

When starting klipper, I keep getting the "Unable to read tmc UART 'stepper_a' register IFCNT" error, and I just can't figure out where is goes wrong... I checked if the analog pins are defined in the pins.py file, and they are. The only thing I can think of is that klipper does not support UART over analog pins...

I also tried the standard ramps uart pins but to no avail...

Perhaps you guys see something I missed! Any help is greatly appreciated!

Datasheets MKS Gen L V2.0 (in product description):

https://nl.aliexpress.com/item/1000006974406.html?spm=a2g0s.9042311.0.0.71004c4d2lvPqi

[klippy.log](https://github.com/KevinOConnor/klipper/files/3913175/klippy.log)

theopensourcerer, 2019-12-02:
I have the MKS Gen v1.4 and use TMC2208s in UART mode on X, Y and Z without problems. You only need a single wire for each uart connection. Here is my config:

https://paste.ubuntu.com/p/RFJ8xCS8MG/

holmlrs, 2019-12-02:
I see. I tried your uart pins, but as expected that did not work.

The MKS Gen L V2.0 has hardwired UART lines, connected directly to PK1 - PK4 (analog9-analog12)

theopensourcerer, 2019-12-02:
Maybe someone else has this board then... But I am not sure at all about that analog pin numbering... If they really are analog pins on then they can't do UART...

holmlrs, 2019-12-02:
They are interrupt pins, so theoreticaly.... Just dont know how klipper handles them (and if it can).

jakep82, 2019-12-02:
You need to specify the digital pin names for UART.  Analog9 is digital pin 63, which translates to ar63 in Klipper.

This is the best image I could find with translations.
https://jgauroraforum.com/uploads/editor/bq/ddnk7mv25bdw.jpg

holmlrs, 2019-12-02:
You are an absolute hero. Works like a charm now! Thank you so much!

closed 2019-12-02 by holmlrs

oft55, 2020-04-05:
Hi @holmlrs, i have a too this problem. Can u upload your configuration? And Which pins did you put a jumper on? take a photo plesea :)

Marcus9812, 2020-06-17:
@oft55 Did you fix them? I have the same problem