Klipper3d/klipper #2240: [Feature Request] MCP23017-E/SO i2c GPIO Expansion Support

jalanjarosz, 2019-12-04:
I have a display the adapter that has a MPC23017-E/SO i2c GPIO expander designed for use on a Re-ARM MCU.  This adapter allows for remapping Re-ARM pins generally used for the display for use as Stepper Driver Cs Pins and using the MPC23017 pins to drive the the display.

Marlin FW supports this chip for the PANELOLU2 displays and for the Panucatt's Vicki displays.

Vicki schematic:  https://s3.amazonaws.com/cdn.freshdesk.com/data/helpdesk/attachments/production/1015336002/original/viki_lcd_schematic.pdf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJ2JSYZ7O3I4JO6DA%2F20191204%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20191204T021824Z&X-Amz-Expires=300&X-Amz-Signature=24e6ac6809e5d80873abec655b10133bcc06dfacfacddf9db1226db6bfbbc05f&X-Amz-SignedHeaders=Host&response-content-type=application%2Fpdf

data sheet for the chip: https://www.mouser.com/datasheet/2/268/20001952C-1129816.pdf

I don't know how similar this is to the SX1509 GPIO expander https://github.com/KevinOConnor/klipper/pull/767 that @FHeilmann implemented.

I'm able to test code and provide feedback as needed.

Thanks!

jalanjarosz, 2019-12-04:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3919679/klippy.log)

FHeilmann, 2019-12-04:
Sorry for not replying to your eMail, I looked into this a bit and here is what I found:

The mentioned LCDs use the aforementioned MCP23017-E i2c to GPIO expander to drive a (I guess) standard HD44780 in 4 wire mode. This is achieved via the LiquidCrystal_I2C library. 

What that library does it takes advantage of the fact that you can set the pins of the expander with one single i2c message, and thus translates the i2c data-stream to a 4-bit parallel signal.

So what would be needed to implement this with klipper:

One would need a way to talk to the MCP. This, as @jalanjarosz already indicated, should not be that different from the SX1509 extra. One would simply have to figure out the appropriate registers and a proper initialization sequence to get the MCP up and running.

What has to happen in addition is some kind of glue that implements the 4-bit-parallel over i2c. As it is now, klipper would bitbang all the LCD pins individually which would likely overwhelm the MCU and the i2c bus.

This brings me to the main concern with this LCD: As it currently stands, the i2c code in klipper is somewhat fragile, as it doesn't handle noise well and doesn't support retries. Furthermore, updating a 128x64 LCD over an i2c bus is likely going to generate a lot of load on the MCU. @jalanjarosz mentioned that he intends to run this on the rearm, so it might not be a huge issue, but definitely something to think about.

In summary, @jalanjarosz, I think if your intention is to save pins, my recommendation is to look into the other display types that do not require a ton of pins. Klipper supports a couple of display types and some of them can be controlled, e.g.,  via SPI, which only requires CLK, MISO, MOSI and a chip-select.

Bests,
-Florian

jalanjarosz, 2019-12-04:
@FHeilmann Thanks for your review and feedback.

For reference, in case there is someone who is willing to implement a library for the MCP23017, I'm attaching an image of the prototype board and the schematic.

[LCD144-B-JJ.pdf](https://github.com/KevinOConnor/klipper/files/3921899/LCD144-B-JJ.pdf)

[image]

I'm also have a Fysetc Mini12864 as a display option, which looks like it supports SPI for the display and a few pins for the encoder: https://wiki.fysetc.com/Mini12864_Panel/  
Having MCP23017 supported would still be helpful to reduce the octopus of jumper wires, as the SPI GPIO would be for the encoder data, and not the display data

I'll also be investigating other options for wiring TMC Cs pins as a backup option.

Thanks for your consideration.

jalanjarosz, 2019-12-04:
@FHeilmann I'm working with MrAlvin on his design and he mocked up another possible solution.  If we drive the RRD display off the MCU pins, like is typically done, but off-load the encoder pins to the MCP23017 GPIO expander, would that reduce the concern/issue of i2c bus noise?

Here is an updated schematic: 
[Was LCD144-B is now LCD144-C.pdf](https://github.com/KevinOConnor/klipper/files/3922343/Was.LCD144-B.is.now.LCD144-C.pdf)

Basically, on the EXP-1 display, the EXP1-3, EXP1-4, and EXP1-5 would still be going directly to the MCU for display updates.

We would still have the 5 needed MCU pins to support Cs pins for TMC drivers.

KevinOConnor, 2019-12-04:
If you need additional gpio pins, I would recommend taking a look at the stm32f042 and stm32f070 chips.  These chips are low cost, and can be directly routed to the host machine via USB.  (I'm also told that the tsop20 versions of the chip can be hand soldered.)

Alternatively, if pcb layout and soldering aren't desirable, one can wire an arduino nano to a display via some jumper wires.  Again, one can then directly control the additional mcu via USB.

-Kevin

closed 2019-12-06 by jalanjarosz

jalanjarosz, 2019-12-06:
I'm going to close this feature request for now.

I added in a Re-ARM as a second MCU to my test system and wired a Fysetc MKS Mini v2.1 display directly into the J3 and J5 headers on the Re-ARM and the display, encoder and NeoPixels work perfectly.   We are going to investigate extending up the J2 and J5 for display usage and extending up the J12 for TMC Cs usage.

For anyone curious how the Fysetc display is configured, here is that section of my Klipper config. Note that the Re-Arm is a second MCU with the prefix "rearm".

```
# Re-ARM board does not have an option to designate SPI pins via config options.
# SPI pins were wired from Re-ARM headers J3 to display as follows:
# Fysetc MKS Mini v2.1 Pin ~ Re-ARM MCU Pin ~ Re-ARM Header ~ RAMPS v1.4.4 pin 
# EXP2-1 ~ P0.17 ~ J3-6 / MISO ~ D50
# EXP2-2 ~ P0.15 ~ J3-9 / SCK ~ D52
# EXP2-6 ~ P0.18 ~ J3-10 / MOSI ~ D51
# Fysetc MKS Mini Display v2.1 works on Re-ARM when wired with the following

[display]
lcd_type: uc1701
cs_pin: rearm:P1.23 
a0_pin: rearm:P1.21
rst_pin: rearm:P1.31
contrast: 63
encoder_pins: ^rearm:P3.26, ^rearm:P3.25
click_pin: ^!rearm:P2.11
kill_pin: ^!rearm:P1.22

[static_digital_output beeper]
pins: rearm:P1.30

[neopixel mks_display]
pin: rearm:P2.6
chain_count: 3
color_order_GRB: False
initial_RED: 0.4
initial_GREEN: 0.0
initial_BLUE: 1.0
```

[image]