Klipper3d/klipper #2241: G1 Z0 leaves nozzle 0.75mm too high after PROBE_CALIBRATE

ZathrasOne, 2019-12-04:
I followed the instructions on GITHUB for calibrating the BLTouch Z offset using the paper test method and PROBE_CALIBRATE, but when I do a G1 Z0 after, the nozzle is 0.75mm too high. I have done it a couple of times with the same results. The BLTouch consistancy test gives a range of 0.0.032500 and 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3920162/klippy.log)

a 0.008231 standard deviation. I have attached my klippy.log and printer.cfg files.

ZathrasOne, 2019-12-04:
Printer.cfg file added with extension change.
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3920173/printer.cfg.txt)

mattkenn4545, 2019-12-05:
Looks like you need to set a z-offset ie

`[bltouch]
sensor_pin: ^PC4
control_pin: PA4
x_offset: -51
y_offset: -18
z_offset: 2.015
speed: 5.0`

ZathrasOne, 2019-12-06:
I thought that is what the PROBE_CALIBRATION routine does?

BlackStump, 2019-12-06:
check your homing override. did you make it yourself? or copy and paste someone elses?
have a read of this and make sure you need those options and what they cause/do
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#force-movement
also look as 
https://github.com/KevinOConnor/klipper/blob/master/config/sample-probe-as-z-endstop.cfg
and decide if you need a homing override or is safe home a better option for you.

KevinOConnor, 2019-12-23:
I'm going to close this issue as it appears the conversation has concluded.  As noted, the best advice is to look closely at the homing_override configuration.

-Kevin

closed 2019-12-23 by KevinOConnor