Klipper3d/klipper #2249: sending parameters to text only gcode macro names returns "Malformed Command"

ld3300, 2019-12-07:
I am having trouble with sending parameters to gcode_macro names that do not end in a number. 
My test gcode macros:
```
[gcode_macro NUM_EXTRUDERS]
default_parameter_S: 2
gcode:
    M117 {S}
    RESPOND TYPE=echo MSG="S{S}"

[gcode_macro NUM_EXTRUDERS1]
default_parameter_S: 2
gcode:
    M117 {S}
    RESPOND TYPE=echo MSG="S{S}"
```
In the Octoprint terminal if I enter `NUM_EXTRUDERS` or `NUM_EXTRUDERS1` I get the expected response: 

> Recv: echo: S2

Also, if I enter `NUM_EXTRUDERS1 S5` I get the expected response:

> Recv: echo: S5

However if I enter `NUM_EXTRUDERS S5` I get:

> Recv: !! Malformed command 'NUM_EXTRUDERS S5'

It appears that since g-code does not distinguish between commands with spaces (i.e. G1 Z10 same as G1Z10) that since my desired gcode macro does not end in a number the parser can't distinguish it from a valid command when a parameter is added when called.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3934668/klippy.log)

ld3300, 2019-12-07:
okay, I see that the command `NUM_EXTRUDERS S=5` works.  It could be helpful to add a section to either the G-Codes.md or exampe_extras.cfg that includes rules and guides for macro names and usage.  Additionally a gcode_macro cannot have any numbers in the name unless they are at the end (unless this is a bug instead of a naming rule).  I'm an idiot at Python, so can't contribute as much as I would like, but if I can help by updating the documentation I would be happy to do so.  Are there any other naming restrictions to be aware of?

KevinOConnor, 2019-12-23:
Okay, I'm going to close this as it appears a solution was found for the original issue.  Yes, due to the funky nature of traditional g-code commands, the "enhanced commands" are limited in their use of numbers in the name.  I've added it to my list of things to document, but if someone wishes to submit a PR that's great.

-Kevin

closed 2019-12-23 by KevinOConnor

ld3300, 2019-12-24:
I will do the documentation when I get a few moments.  It looks like it should be added to docs/G-Codes.md and maybe example-extras.cfg.  Are there any other formatting rules or illegal characters that you know of?

JohnEdwa, 2019-12-24:
#2307, config settings gets converted to lower case, so upper case variables don't work.