Klipper3d/klipper #2250: MCU_stepper instance has no attribute 'add_to_endstop' after latest update

hypermind30, 2019-12-07:
My clone BLtouch keeps failing so i decided to update to latest klipper version and after updating i encounter above error. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3935036/klippy.log)

ld3300, 2019-12-07:
Try this from example_extras.cfg

```
#If using "probe:z_virtual_endstop" then do not define a position_endstop in
# the stepper_z config section.
```

.

hypermind30, 2019-12-07:
tried it. got the same error
[klippy2.log](https://github.com/KevinOConnor/klipper/files/3935627/klippy2.log)

ld3300, 2019-12-08:
Now I am wondering if you found some issue caused when the bltouch and z are both not on main cpu.
Though, you are on a dirty build, so maybe updating to 0.8.0 clean would fix it. I would try that first.

BlackStump, 2019-12-08:
I run a bltouch on a secondary mcu for 3 Z steppers with no problems

ld3300, 2019-12-08:
Then all that I can contribute is trying to get to a clean update.  Maybe one of the source files is stuck at a previous version giving the dirty build.

hypermind30, 2019-12-08:
i just tried 
klipper git pull
make menuconfig
make clean
make
sudo service klipper stop
make flash FLASH_DEVICE=/dev/ttyUSB0
make flash FLASH_DEVICE=/dev/ttyUSB1
sudo service klipper start

i still have Firmware version: v0.8.0-174-g72161d0-dirty
and still have the same error 

how do i make a clean update?

hypermind30, 2019-12-08:
ld3300  your the man.
I deleted klipper directory 
and followed klipper installation and everything is working now.
Thank you so much.

closed 2019-12-08 by hypermind30