Klipper3d/klipper #2252: Underextrusion after retraction (BMG, DirectDrive, Ender3)

grykom, 2019-12-08:
Hi guys, 
I have a problem and I'm slowly losing hope.
The situation that I describe only happens in the DirectDrive setting. I never had such a problem with Bowden. My setup is Ender 3 with BMG (trianglelab clone), SKR 1.3 (TMC2208 / TMC2209), stepper motor on extruder is Creality 42-34  (the one from the X and Y axes).
I describe the problem as Underextrusion after retraction and looks like this:
[image]
This happens only after retraction and I have probably tried everything. Different settings for retraction, speed, presure_advance.
Do you have any idea how to fix it? My [klippy.log](https://github.com/KevinOConnor/klipper/files/3935749/klippy.log)

ld3300, 2019-12-08:
Can you reproduce it and do a dump_tmc stepper=extruder when you see it happen?

BlackStump, 2019-12-08:
A BMG is not a direct drive it is still a 3:1 gear reduction, so that might be the first part to look at.
Check the BMG gears are not binding, some of the Chinese BMG clone build quality is pretty iffy.
ie poor fitting bearings, binding gears, just to name a couple that I have seen personally.
Check the bowden tube that goes from the BMG down the throat of your hotend is the correct length.

Strahler85, 2019-12-08:
Is the Extruder Motor Driver a TMC2208/09?
TMC2208/09 connected to UART?
Does the motor driver runs in stealthcop mode?

Hywelmartin, 2019-12-09:
@Strahler85 
[tmc2209 extruder]
uart_pin = P1.8
microsteps = 16
run_current = 0.580
hold_current = 0.580
stealthchop_threshold = 0
interpolate = true

closed 2019-12-10 by grykom