Klipper3d/klipper #2253: [Q] how 'ZAdjustHelper.adjust_steppers' does work?

JYKeith123, 2019-12-08:
Hi I'm very new to klipper code.
I'd like to commit code about 'quad_gantry_level.py',
but I don't know nothing of other codes, coding conventions, and history. 
I changed mathmatical algorithm of 'quad_gantry_level.py' and created own adjust_steppers.
But after development finished,
I figured out 'quad_gantry_level.py' had its own adjust_steppers before originally,
and it moved & merged into 'z_tilt.py'.
The reason I created new 'adjust_steppers', I didn't understand why it works like that.
cause for me, if the function is
'def adjust_steppers(self, adjustments, speed):'
and the parameter 'adjustments' is [-1, 1, 0, 1], 
I would expect the function will work like

> Z0 stepper go down -1mm
> Z1 stepper go up 1mm
> Z2 stepper do nothing
> Z3 stepper go up 1mm

but currunt master version doesn't work like this.
I don't know other codes and what intended so I'm asking it here.
I think each z steppers should response of each value of list.

currunt master version of ZAdjustHelper.adjust_steppers in 'z_tilt.py'

```
def adjust_steppers(self, adjustments, speed):
...
toolhead.flush_step_generation()
for s in self.z_steppers:
	s.set_trapq(None)   # previously it was like s.set_ignore_move(True)
...

for i in range(len(positions)-1):
	...
	toolhead.flush_step_generation()
	stepper.set_trapq(toolhead.get_trapq())
	...
...
toolhead.flush_step_generation()
last_stepper.set_trapq(toolhead.get_trapq())
...

```

I think correct algorithm and original code in 'quad_gantry_level.py' worked like this.
<pre>
<code>
def adjust_steppers(self, adjustments, speed):
...
toolhead.flush_step_generation()
for s in self.z_steppers:
	s.set_trapq(None)   
...

for i in range(len(positions)-1):
	...
	toolhead.flush_step_generation()
	stepper.set_trapq(toolhead.get_trapq())
	...
	<b>stepper.set_trapq(None)</b>
...
toolhead.flush_step_generation()
<b>for s in self.z_steppers:
	s.set_trapq(toolhead.get_trapq())</b>   
...
</code>
</pre>

can someone explain about this?
why first stepper should be adjusted by all 4 values of 'adjustments'
and next one 3, next 2, and last z3 1 value?
is this a bug or intended?

KevinOConnor, 2019-12-08:
> 'def adjust_steppers(self, adjustments, speed):'
and the parameter 'adjustments' is [-1, 1, 0, 1],
I would expect the function will work like

    Z0 stepper go down -1mm
    Z1 stepper go up 1mm
    Z2 stepper do nothing
    Z3 stepper go up 1mm

That is effective what that code ultimately does.

> why first stepper should be adjusted by all 4 values of 'adjustments'
and next one 3, next 2, and last 1 value?

It is done that way because adjusting just one stepper in isolation could put significant stress on the gantry and/or bed.

For example, consider the following adjustments: [20, 25, 0, 0].  If the first stepper adjustment was made in isolation it could result in the second stepper being significantly skewed from the three other steppers and potentially stress the frame.  Instead, in this example, the `adjust_steppers()` code would first move the second stepper 5mm, and then move the first and second steppers the remaining 20mm in parallel.

-Kevin

JYKeith123, 2019-12-09:
Then I think the current ZAdjustHelper.adjust_steppers has a bug.

The latest version doesn't work like that you mentioned.
To test, I added some code for debugging like below
<pre>
<code>
    def adjust_steppers(self, adjustments, speed):
        ...
        for i in range(len(positions)-1):
            stepper_offset, stepper = positions[i]
            next_stepper_offset, next_stepper = positions[i+1]
            toolhead.flush_step_generation()
            stepper.set_trapq(toolhead.get_trapq())
            <b>gcode.respond_info('start height : {}'.format(curpos[2]))</b>
            curpos[2] = z_low + next_stepper_offset
            <b>gcode.respond_info(
                'stepper: {}, curpos: {:.6f}, z_low: {:.6f}'.format(
                    stepper.get_name(),
                    curpos[2],
                    z_low
                )
            )
            continue</b>
            try:
                toolhead.move(curpos, speed)
                toolhead.set_position(curpos)
            except:
                logging.exception("ZAdjustHelper adjust_steppers")
                toolhead.flush_step_generation()
                for s in self.z_steppers:
                    s.set_trapq(toolhead.get_trapq())
                raise

        # Z should now be level - do final cleanup
        last_stepper_offset, last_stepper = positions[-1]
        toolhead.flush_step_generation()
        last_stepper.set_trapq(toolhead.get_trapq())

<b>        gcode.respond_info(
            'stepper: {}, curpos: {:.6f}'.format(
                last_stepper.get_name(),
                curpos[2] + first_stepper_offset
            )
        )

        '''curpos[2] += first_stepper_offset
        toolhead.set_position(curpos)
        gcode.reset_last_position()'''</b>
</code>
</pre>

> try 1. ZAdjustHelper.adjust_steppers, input [3, 0, 0, 0]
> result:
> start height : 30.0
> stepper: stepper_z, curpos: 33.000000, z_low: 33.000000
> stepper: stepper_z1, curpos: 33.000000, z_low: 33.000000
> stepper: stepper_z2, curpos: 33.000000, z_low: 33.000000
> stepper: stepper_z3, curpos: 30.000000
z0 stepper go up 3mm
all stepper go down 3mm

> try 2. ZAdjustHelper.adjust_steppers, input [20, 25, 0, 0]
> result:
> start height : 30.0
> stepper: stepper_z1, curpos: 35.000000, z_low: 55.000000
> stepper: stepper_z, curpos: 55.000000, z_low: 55.000000
> stepper: stepper_z2, curpos: 55.000000, z_low: 55.000000
> stepper: stepper_z3, curpos: 30.000000

z1 go up 5mm 
z0, z1 go up 20mm
z0, z1, z2 no move
z0, z1, z2, z3 go down 25mm

> try3. ZAdjustHelper.adjust_steppers, input [-1, 2, -1, 3]
> start height : 30.0
> stepper: stepper_z3, curpos: 31.000000, z_low: 33.000000
> stepper: stepper_z1, curpos: 34.000000, z_low: 33.000000
> stepper: stepper_z, curpos: 34.000000, z_low: 33.000000
> stepper: stepper_z2, curpos: 31.000000

z3 go up 1mm
z1, z3 go up 3mm
z0, z1, z3 no move
z0, z1, z2, z3 go down 3mm

total move
z0 go down 3mm
z1 no move
z2 go down 3mm
z3 go up 1mm

as you can see, try 1 & 2 work correctly but try 3 doesn't
and also while doing this I figured out that 'last_stepper_offset' is not used.

Now I know what intended to do originally.
Can I fix it and put it on my 'quad_gantry_level' pull requests?

BTW, from 'quad_gantry_level' perspective, I thought the situation prevented by below.

> 1. Set adjustment value on average based to reduce absolute value
> 2. QuadGantryLevel.max_adjust limits maximum value

KevinOConnor, 2019-12-09:
> total move
z0 go down 3mm
z1 no move
z2 go down 3mm
z3 go up 1mm

That is not correct - z3 went up 4mm and z1 went up 3mm - the remaining manipulations were via `toolhead.set_position()` which did not physically move the toolhead.

There are test cases in test/klippy/z_tilt.test and test/klippy/quad_grantry_level.test which can be run to demonstrate how this process works.  See the docs/Debugging.md document for info on running the test cases.  (To run a particular test case with output, use something like: `~/klippy-env/bin/python ~/klipper/scripts/test_klippy.py -d dict/ ~/klipper/test/klippy/z_tilt.test -v`

> Can I fix it 

If you've experienced a problem with the code, you'll need to start by describing the problem and providing the klipper log file of the session that produced the error.  See: https://www.klipper3d.org/Contact.html

-Kevin

JYKeith123, 2019-12-09:
Thanks for explanation

closed 2019-12-09 by JYKeith123