Klipper3d/klipper #2254: FIlament Change - RESUME failed, Klipper reports 'resumed' but nozzle stuck in one position

drelich, 2019-12-08:
Hi,

I tried some filament change magic today and it 'sort of works'. Filament changes invoked by the actual gcode worked (printing off the virtual SD card). Had a cube with three filament changes, and they worked fine.

However, when I tried to change filament the fourth time and using my Macro buttons, I can change the filament no problem but the RESUME function fails - it does move the nozzle back to where it left off but it remains in that position, and the Terminal shows it's printing the same byte off the SD card over and over again... 

Any help/fix would be greatly appreciated. Cheers.

David.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3936058/klippy.log)

Arksine, 2019-12-09:
It appears you are on a forked repo.  You'll need to pull the latest version of this repo and reproduce the issue for us to help.

drelich, 2019-12-09:
I pulled that repo only a few days ago. No matter, I'm back to Marlin. I can't afford wasting any more filament on failed prints. Cheers.

closed 2019-12-09 by drelich