Klipper3d/klipper #2257: BlTouch fails after 5 probes

tonn0297, 2019-12-08:
hello kevin, 
i have a issue at hand that i can't figure out. 

I'm running 2 boards.
1. board controls x-y-e, Fans, and heaters
2. board runs the z and bltouch v3.0 (3.1 same issue, difrent pins same issue)

when i have probed like 5 times the bltouch goes in error state. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3936581/klippy.log)

ld3300, 2019-12-09:
It looks like the BL-touch is either not reading itself correctly or the connection from the BL-touch to the Z board isn't solid.  Did you try adjusting the screw on top of the BL-Touch to adjust the trigger location?

w1ebr, 2019-12-09:
Does it always fail on the fifth probe or are the failures occurring often but seemingly randomly? I ask because I have also had this problem with a Ver. 3.1 ( it fails trying to retract the pin) and my solution  at the moment is to change the bltouch code so that the retry issued on line 108 includes a duration option so that it waits long enough for the pin to go up before it checks to see if the bltouch was triggered. This allows my bed leveling to continue

tonn0297, 2019-12-09:
the wiring is okay. ive messures it thrugh and i have not changed any hardware. only added an extra board.

tonn0297, 2019-12-10:
@KevinOConnor i can't tell if theres anything that drifts, clocks timing or somthing like that. maybe you can? all wires are okay

KevinOConnor, 2019-12-13:
Is it the same issue discussed at https://github.com/KevinOConnor/klipper/issues/1938#issuecomment-529255057 ?

-Kevin

tonn0297, 2019-12-13:
yes and ive tried some of the fixes in that thread with no luck. 
-Tonny

KevinOConnor, 2019-12-16:
At https://github.com/KevinOConnor/klipper/issues/1938#issuecomment-529646345 you indicated the work around was successful.  Unfortunately, a real fix has not been implemented yet.  I can push forward the temporary work around if that is needed.

-Kevin

tonn0297, 2019-12-16:
it do work on the board i had issues with at that time, but now i have it on my 2nd printer and that's where i have the issues right now. 
A fix would be very appreciated rn. 

-Tonny

zmrde, 2019-12-19:
I am on the same boat and many others too.

tonn0297, 2019-12-20:
@KevinOConnor is there any updates on this? we are 3 and counting that have issues with the unstable probe. 

-Tonny

KevinOConnor, 2019-12-23:
Your log indicates that the Klipper micro-controller detected the BL-Touch signal, stopped the motors, relayed the info to the Klipper host software, and then implemented a "pin_up" command in ~106ms from the event.  So, Klipper is operating normally.  It would seem that the BL-Touch entered into an error state anyway.  It is not known why.  

Klipper could have done a better job at attempting to clear the error state - see: https://github.com/KevinOConnor/klipper/issues/1938#issuecomment-529736513

I don't think we need to track this separately from the existing issue #1938 .

-Kevin

KevinOConnor, 2020-01-03:
I'm going to close this issue in favor of the existing issue #1938 .

-Kevin

closed 2020-01-03 by KevinOConnor